betaflight/betaflight #2398: EXTI_IRQ bug of F4 MPU_INT on PB pins

MATEKSYS, 2017-02-14:
If put the MUP_INT on one of PB pins of F4, & enable the EXTI_IRQ. The F4 system halted.
MPU_INT on PA & PC pins, no problem.

https://github.com/betaflight/betaflight/blob/master/src/main/drivers/exti.c
210 #if defined(STM32F1) || defined(STM32F7)
211 _EXTI_IRQ_HANDLER(EXTI2_IRQHandler);
212 #elif defined(STM32F3) || defined(STM32F4)
213 _EXTI_IRQ_HANDLER(EXTI2_TS_IRQHandler);

Move the "|| defined(STM32F4)" to line 210. Bug is solved.

ledvinap, 2017-02-14:
+1

Current IRQ handling code is quite easy to break, there is no check that handler name is spelled correctly.
It should be possible to use macro and `enum IRQn` to validate handler name ... I'll look into it

ledvinap, 2017-02-14:
https://github.com/cleanflight/cleanflight/pull/2619

Should be trivial to rebase on betaflight ...

ledvinap, 2018-05-07:
worked on in #2404, but eventually stalled ...

closed 2018-05-07 by ledvinap