betaflight/betaflight #2403: Racefight Revolt V2

rcquad808, 2017-02-14:
Has anyone used RF Revolt V2 board with Betaflight.  In the Receiver tab it won't see the transmitter, everything else works. The V2 has a different wiring than the V1 version. UART1, 3, and 6 are the only one in the tab.  The Sbus is wired to the TX1 pin on the V2 board and the V1 it was wired to the SBUS pin.  I take it that the V2 board is not yet support by Betaflight?

jflyper, 2017-02-15:
Your Revolt V2 will be resurrected with v3.1.6 patch release (1st, March).

If you don't wanted to be grounded for two weeks, then go take a look at the BF wiki, under "Boards /  REVOLT 2".

closed 2017-02-15 by blckmn

rcquad808, 2017-02-15:
Hello,

Nice! i can wait for it.

Thank you!

AndersHoglund, 2017-02-15:
Why wait? Try it now and be a part of early testing.  
CI build available here: http://andwho.sytes.net:8080/job/BorisB_BetaFlight_Maint/

rcquad808, 2017-02-15:
Hi,
So is that where i can download the firmware to test Betauon revolt 2?

Erol

turbosi92, 2017-02-15:
jflyer mentions above what you need to do.  ;)

AndersHoglund, 2017-02-16:
The link in the WiKi points to a private build based on some state of BF3.2.0 
My link above is the current state of BF3.1.6 . Your choice.

blckmn, 2017-02-16:
We should update to the 3.1.6

I only did that build for testing - I would not recommend it for widespread use.

rcquad808, 2017-02-16:
Ok thank you