betaflight/betaflight #2405: betaflight wont start on my computer

multirotorbill, 2017-02-15:
for the last 2 weeks I cant get Betaflight to run on my computer. I had been using it, then one day Betaflight would not run.. cleanflight runs fine , and to this day I still cant get betaflight to run ... I really do prefer to run on betaflight but I need to fly , so cleanflight it is until I can get Betaflight to work again...

andymac4182, 2017-02-15:
Have you tried removing and installing the chrome app again?

multirotorbill, 2017-02-16:
Yes I have completely uninstalled chrome and used the chrome clean up tool and then restarted my. Computer and reinstalled chrome..and no joy..

mikeller, 2017-02-16:
@multirotorbill: Can you check what errors the Betaflight configurator is throwing? In Chrome, go into the menu / 'More Tools' / 'Extensions'. Then find the Betaflight configurator and click 'Launch'. There should be an 'Errors' link. Click it, and copy / paste what appears there.

nathantsoi, 2017-02-16:
maybe we should get this merged in:
https://github.com/betaflight/betaflight-configurator/pull/316, def. will
help w/ cases like this

mikeller, 2017-02-16:
Yes, https://github.com/betaflight/betaflight-configurator/pull/316 will be a good addition once it works.

multirotorbill, 2017-02-17:
ok, not sure why but, I went to menue,more tools, ext, tried to click on betaflight nothig happens.. So obviously I dont know what Im doing here because those were some pretty clear and straight forward instructions....I dunno, I will keep trying to get an error report.

closed 2017-10-23 by jflyper