betaflight/betaflight #2407: Blackbox not working on Lux v2

fpvacademy, 2017-02-15:
Hey guys!

Blackbox worked fine on the stock BF 3.0 that the Lux v2 shipped with, but as soon as I updated it to all other BF versions, I get a fatal Error for Blackbox. I tried everything to get it working and I believe that this has to be a Firmware issue.

mikeller, 2017-02-15:
Can you be a bit more specific about what configuration you are using, and how this 'fatal Error' manifests please?

fpvacademy, 2017-02-15:
The fatal error shows up on the Blackbox tab. The SD Card icon is red.

I have tried literally all configurations possible. Spent about 2 hours trying different variations. I'm no BF noob either. No matter what configuration I try, the error still persists.

mikeller, 2017-02-15:
Ah I see. There is a problem with DMA for SDCard that affects some boards. Testing on a LUX V2, after setting `set sdcard_dma = off` the SD Card reader works fine.

fpvacademy, 2017-02-15:
Just tried that. Didn't fix it for me. What config and version of BF did you use?

mikeller, 2017-02-15:
3.1.5, default config, only thing i changed from default was `set sdcard_dma = off`, and it worked. Have you tried reformatting your SDCard with the official formatter (https://www.sdcard.org/downloads/formatter_4/)?

fpvacademy, 2017-02-15:
Just tried it exactly like that. No fix.

This doesn't make sense to me. I've used this SD Card many times with SP Racing Evo Blackbox before and it works fine. It also works fine with Stock custom 3.0 that Lux v2 shipped with, but can't get it to work with 3.1.5 or anything or than stock FW on Lux v2.

mikeller, 2017-02-15:
I just tried again with a LUXV2_RACE, and stock 3.1.5. Both with default settings and with `set sdcard_dma = off` the SDCard is recognised just fine, and can be written to by blackbox.

fpvacademy, 2017-02-15:
So weird, got it working on my secondary quad, but not primary.

Now the problem is that when I record Blackbox and I want to open to view, I get the error: "Log is truncated, contains no data"

Any ideas?

mikeller, 2017-02-15:
You might be running too high loop speeds, causing writes to be dropped. Try very conservative loop speeds to see if that fixes it.

fpvacademy, 2017-02-15:
Lowered loop and log speeds. Still nothing. Other suggestions?

mikeller, 2017-02-15:
The only one that comes to mind is: Faulty board. If the same setup works on a different board (and on my board as well), then the original board is faulty.

borisbstyle, 2017-02-19:
I have 2 LUXV2 FC's over here and on both it works just fine!

fpvacademy, 2017-03-02:
Hey guys! This problem still persists. The LUX V2 works fine; it reads the card and writes data to it, however, when I want to open the file in BlackBox Explorer, it tells me that the "Data is Truncated".

So it will show fine on your LUX whenplugged in and not connected to a quad, but as soon as you want to record, it doesn't work. :(

Any ideas? Thoughts?

mikeller, 2017-03-02:
@fpvacademy: Have you tried with a different board? From your description it could well be that the board in question _appears_ to be writing fine, but the data is garbled because of hardware damage.

fpvacademy, 2017-03-03:
I'm such a tool. I've been trying to open it in CF BlackBox Explorer. Didn't realise there was one for Betaflight. All works fine. Thanks for all your help and hard work guys!

closed 2017-03-03 by martinbudden