betaflight/betaflight #2409: Micro Minim OSD does not work CC3D Revolution BF3.1.5

rolial, 2017-02-15:
Hello everyone!
I'm having trouble making Micro MinimOSD work with FC CC3D Revolution.
I'm using:
Betaflight 3.1.5
RX  Spektrum Satellite in  Fex Port;
GPS in Main port;
Micro Minim OSD on the FlexiIO port (MW OSD R1.6)

Everything is working except the OSD.
I made the configuration in MWOSD for Betaflight.
I am using the wires in this sequence (without connecting the black and red battery cable on the bottom left).
Am I doing something wrong?
[image]

mikeller, 2017-02-15:
Have you configured the UART that the MWOSD connects to to have MSP enabled?

rolial, 2017-02-16:
Yes, I did. Look at the picture..
What I'm finding strange is the micro minimOSD led is blinking red not green as I've seen in the MW OSD setup tutorials.
Could anyone send a MW OSD link already ready for Betaflight 3.1.5 and FC CC3D Revolution for me to test?
Are the wires connected right?
Look at the  photos ...
[image]
[image]