betaflight/betaflight #2412: Increase failsafe_off_delay range

rfengfpv, 2017-02-15:
In BF 3.1.5 the maximum value allowed for failsafe_off_delay is 200 (according to the CLI message received when trying to enter a value greater than 200, although the GUI accepts values up to 255). This corresponds to 20 seconds. At the recommended descent rate of 1 m/s this is only sufficient for a descent of 20m. For quads that are flown higher than 20m AGL it would be nice to be able to set a larger value. My recommendation is that values up to 600 (60s) be permitted. If the range is due to storage as a byte, then one could consider making it the time in seconds, not in tenths of a second, since there seems no reason why the failsafe_off_delay needs to be set with better than 1s precision.

mikeller, 2017-02-15:
@rfengfpv: The reason it is restricted to 255 is that it is transmitted / kept in an 8 bit variable indeed. I am a bit reluctant to change the definition from decaseconds to seconds, since people confusing this will have unintended and bad consequences. But what could be done is to change the configuration variable to 16 bit, and making the full range available in the CLI. Would this help for your use case?

Also, when you are talking about 60s `failsave_off_delay`, this makes me wonder if this isn't more of a use case for iNav anyway, since an uncontrolled craft without GPS position hold taking 60s to land will mean that it will have the potential to drift for a long distance during the time it takes to land, even with moderate wind.

rfengfpv, 2017-02-15:
Hi Michael

Thanks for your response. I agree about the danger of unintended
consequences if the definition is changed. A 16 bit variable with
decasecond precision would be a good solution. I personally would not set
60s - the value I actually wanted to set was 30s since I don't often fly
higher than 30m/100ft (our max legal limit is 150 ft). I had a failsafe
from low altitude today which made me realise that my current setting of
10s was quite insufficient as the motors cut a couple of seconds after
landing!

I'm flying a miniquad without GPS so inav isn't a solution for me. As for
wind, (a) that only applies on windy days, while gravity always applies,
and (b) I've had good luck finding lost quads using a directional antenna
to DF the video signal!

Thanks again,
Andrew

andre92, 2017-02-16:
Miniquad? Just drop it. Much safer to everyone on the ground, instead of getting your face sliced by spinning props. Or think about kids picking it up after it drifts, and has landed with the blades still spinning

rfengfpv, 2017-02-18:
The best option depends on the circumstances. I fly mostly on private land
where there are no other people within the drift range. If it fail-safes at
the top of a power loop 30m/100ft up then I would much prefer that it right
itself and descend at a moderate rate than just dropping it, which could do
all sorts of damage! All I'm asking is that the range of values allowed for
one of the configurations be extended, since ones options should be
determined by the circumstances and not dictated by software limitations. I
don't claim that this is a critical change though - I can certainly live
without it if it's going to cause problems.