betaflight/betaflight #2413: PID new values don't save

Xguy4ku, 2017-02-15:
version 3.15,  PID values revert back to stock when I disconnect reconnect quad to configurator,anyone knows what's up? thanks

mikeller, 2017-02-15:
Hi @Xguy4ku. I can't reproduce that one, PID values save just fine for me after changing and saving them. Are you sure you did not forget to hit 'Save' after changing and before disconnecting?

Xguy4ku, 2017-02-16:
yep, save disconnect , reconnect, they're still there an fine, unplug USB , reconnect back to default.

Xguy4ku, 2017-02-16:
same for two different quads, both running Dodo, one running one shot and the other D-shot

mikeller, 2017-02-16:
Question: Are you using different profiles on the PID tab? What could happen is that the selected profile (shown at the top of the PID tab) changes when you power cycle the flight controller.

Xguy4ku, 2017-02-16:
Apparently I had to save it then go back to the configuration tab and hit save and reboot again for it to work, I did not know that thaks for your help

closed 2017-02-16 by Xguy4ku