betaflight/betaflight #2415: Missing support for WRGB WS2812-Strips (Neopixels)

reptil1990, 2017-02-16:
The RGBW Strip is not supported! The additional color is messing up the channels and the Colors are not displayed correct.
I try to work on a library but would love to get some help!

Thank you!

Carsten Graf

HaseebNain, 2017-02-16:
What seems to be the issue? I just tried the latest build and all colors seem to be supported, even tried custom colors (double-clicking the a color selection in the LED Strip page). 

What FC are you using? Are no colors showing? Is the strip soldered correctly?

MennoBr, 2017-02-16:
I think he's using a ledstrip with RGBW leds 4 colors and not RGB leds 3 colors.

reptil1990, 2017-02-16:
Correct! The Strip has 4 colors Red Green Blue and White. That means an additional channel for each LED. If I am correct one channel contains 8 bit and so the Values are shifted. The first LED is OK but with each additional LED it is getting worse.

mikeller, 2017-02-16:
@reptil1990: Thanks for offering to help with improving Betaflight. The LED output frames are generated here: https://github.com/betaflight/betaflight/blob/master/src/main/drivers/light_ws2811strip.c#L160-L164

I think just adding the additional value for the W channel should be pretty straightforward. Feel free to open a Pull request once you've got something that works, I am happy to work with you on properly integrating it into Betaflight.

reptil1990, 2017-02-16:
I think I can get this working but it would be nice if there will be a possibility to select the type of the strip in the CLI, can someone help with that? Thank you!

mikeller, 2017-02-16:
Sure. That's what I meant with 'integrating it into Betaflight'. :-)

Best way to make it configurable is to add a `ws2811Type` or similar parameter to `ws2811UpdateStrip` in https://github.com/betaflight/betaflight/blob/master/src/main/drivers/light_ws2811strip.c#L141

Then, in `master/src/main/io/ledstrip.c`, set the type according to a new `currentLedStripConfig->ledstrip_is_ws2812` config variable. (You can use `currentLedStripConfig->ledstrip_visual_beeper` as a guideline on how to add a variable.)

reptil1990, 2017-02-18:
I have tried now 2 evenings to get something running and compiled my code often and flashed it to test.. But nothing worked so far.. 
My problem are the additional 8 bits.... can you make an easy fix? I tried it but I think thats above my horizon (not a programmer at all) 
My idea was just to add 8 empty bits for the white color. 
This is the Library of the pixels and the line i think the bits are shifted:
https://github.com/adafruit/Adafruit_NeoPixel/blob/master/Adafruit_NeoPixel.cpp#L85
Thank you so much!

mikeller, 2017-02-18:
Hi @reptil1990. I had a quick look into the different multicolor LED variants that are available. I couldn't find any good documentation on the protocol used for RGBW, and not even good documentation on RGBW being supported by WS2812 LEDs, or only by the SK6812 type, which uses a slightly different timing.

Seeing that there is a whole bunch of different protocols out there, and that there are entire libraries dedicated to generating the signal for the various variants, I don't think that adding support for just one more of these variants is a good move for Betaflight, since space for new features is already very limited for all but F4 based boards.

So my recommendation is to get yourself some WS2811 or WS2812(RGB) based strips, and use your neopixel strips for a different project.

reptil1990, 2017-02-18:
Your right so close this issue and thank you for having a look!
Keep going with the amazing work!

closed 2017-02-19 by mikeller