betaflight/betaflight #2422: Naze 32 rev 6 Full Throttle

Ghostbomb, 2017-02-17:
My quad which has a naze32 rev 6 board running Betaflight 3.1.5 sometimes when i just plug in the battery and try to arm it nothing happens and then it flips over and goes full throttle into the ground. I works fine and I can fly but just sometime this just happens. This has happened to me in the past with earlier versions of Betaflight (2.9.1-3.1.5).  I 3 cells and 4 cells and I think it only does this with 3 cells. I don't remember it ever happening with a 4 cell.

Hardware:
Receiver: Spektrum Quad Race Serial Receiver w/ Telemetry (SPM4649T)
Radio: Spektrum DX8 Gen 2
PDB: Matek Mini Power HUB with BEC 5V & 12V
Motors: EMAX RS2205/ 2300Kv
ESC: Lumenier 30A BLHeli ESC (Running Oneshot125)
Flight Controller: Naze32 Rev 6

fftunes, 2017-02-17:
If it happens across that many versions of betaflight... more likely hardware fault? 
Rx failsafe working correct? Also some matek boards are said to be not very reliable, i heard the 5V is sometimes not "starting" well which could make sense when you say it happens only with 3s batteries.

Hundelunge, 2017-02-18:
i have the same problem with 4s

borisbstyle, 2017-02-20:
what looptime are you running?

Ghostbomb, 2017-02-21:
How do I check my looptime?
Sorry, I'm pretty new to this.  
@borisbstyle