betaflight/betaflight #2426: DSHot low throttle / DShot digital_idle_percent

mjedlinski, 2017-02-17:
Hi.
Recently i build new quad with betaflightF3 and Kiss24re.
I have problem with low throttle "deadband"
Like on picture FC start increasing motors when my throttle rises above 1100.

[image]

Second thing.
I have problem with fast rolls and flips . Flips and rolls end at different times.
On blackbox it look like motor stop and have problem with start to spin. Some similar to loose prop (max motor output without thrust).
I tried to increase digital_idle_percent from 4 to 7 (it's min_throttle to dshot i belive) with no effect.
[image]

AndersHoglund, 2017-02-17:
1. Assuming you have Tx/Rx sticks low end points at 1000 and `min_check` at 1100 (default value), it is kinda expected that throttle will kick in at 1100. Lower `min_check` to where you want Tx throttle stick to become active. 1030 or so is not uncommon, but it is a matter of personal taste.
2. Lack of power at low rpm. If raising `digital_idle_percent` does not help, you have to play with ESC things like timing, de-mag etc etc. Or reduce roll/flip rate to what your motors can handle. You are probably asking more from your gear than it can deliver.

And please, this is not a general tuning help site. It is intended for reporting betaflight firmware issues. You would probably get more tuning help on RCG or FB.

Cheredanine, 2017-02-17:
Also re 2 check the wiki on setpoint

closed 2017-02-18 by mjedlinski