betaflight/betaflight #2427: EEPROM Reset sometimes at initialisastion

Faduf, 2017-02-17:
Hi,
I'm working on a F4 board called YupiF4. Sometimes when I plug the Lipo, the ESC make the initialisation sound 2 times and it is impossible to arm. After unplugging the battery and then reading the configuration on betaflight configurator, I can see that the EEPROM was reset to default.
I saw in the code that the EEPROM can only be reset to default with the MSP_RESET_CONF command or because of a False returned by the function ensureEEPROMContainsValidData. The problem is happening only when I plug a Lipo and without any MSP communication so I don't think that the reset is coming from the MSP_RESET_CONF command. The only possibility is a  False returned by the function ensureEEPROMContainsValidData but I don't understand why. 
Has someone already experienced that problem? How can I troubleshoot this?

blckmn, 2017-02-18:
you would need to debug it and find out why it has reset from the validate function.

MJ666, 2017-02-18:
Could be some bad memory cells in the EEPROM flash area. If checksum is invalid the configuration is reset.

Faduf, 2017-02-19:
I have flown 10 packs yesterday with the same board and the beta 3.1.5. I got no reset. I also think, it could be some bad memory cells, leading to a wrong checksum. I am closing the issue.

closed 2017-02-19 by Faduf