betaflight/betaflight #2430: Desync?/Stall issues after installing BF 3.1.5 on X-Racer 3.1

InsertUserHere, 2017-02-19:
Hi, I basically upgraded the firmware, and the quad randomly desync in middle air during 1 out of 5 fast rolls or flips... I cranked up the min_throttle to 1085 which is already too high, I heard someone saying that I should deactivate Demag compensation if Im constantly using airmode via blheli suite, so I did it.

I noticed that the ppm_min_throttle of my escs in the BLh suite were really low like 1014, Idk if this has something to do.

Just looking for a suggestion or a tip.

llambkin, 2017-02-19:
Yeah, Ive had this once in the last week, But I am flying Omnibus F4 with KISS 24A and frsky XSR. It wasnt a failsafe, as I was only 7odd meters away. And I was in the middle of doing a roll. I was rolling, and I was about to do a reverse roll when it infinirolled into the ground. Hasnt happened since.

mjedlinski, 2017-02-19:
I have similar issue with KISS24A DShot and BetaflightF3.
I rise digitalidleOffsetPercent from 4,5 to 7 and was the same. So I rise it to 9 and now i don't have desync at all, but i think my idle throttle is little to high (than that I'm used to).