betaflight/betaflight #2431: 6 Quads flipping when landing

enc0de101, 2017-02-19:
Hi,

I first noticed this with one of my quads running DSHOT 600. I went to land and once i dropped the throttle it went crazy could of killed someone. So i thought something is wrong with it so i ignored and flew my other same thing but its at random. I have tested this with multishot,oneshot and dshot all have same problem and acro and angle mode with and without airmode. 

Am i missing something in the new updates that i need to configure via CLI.

Thanks in advanced for all the help.

llambkin, 2017-02-19:
Check the wiki and read everything about D-Shot.
https://github.com/betaflight/betaflight/wiki/DSHOT%20ESC%20Protocol
I think the setting you might be looking for is:
set digital_idle_percent =

crash-n-burn, 2017-02-27:
encOde101 I had a very similar issue. Flights where fine, but when I landed my quad, as soon as it touched ground with throttle full off it would twitch, bounce, flip and go hay wire until I disarmed. If this sounds like the problems you have, check your esc's in BLHELI. I activated "damped light" mode and no problems 3 flights/batteries in a row. I kept landing and taking off just to see if it was a fluke. 

This setup 
Insect 130
Omnibus F3 FC
RTF Spider lite esc's PWM mode damped lite activated
DYS 4000kv motors
Lemon Rx DSMX PPM diversity