betaflight/betaflight #2437: OSD feature flag incorrectly set.

hydra, 2017-02-19:
There seems to be a new piece of code that incorrectly enables OSD by default if OSD is compiled in.

See https://github.com/betaflight/betaflight/blob/master/src/main/fc/config.c#L597

IMHO this is wrong and bad, why?  The SPRacingF3NEO has OSD support compiled in but it should NOT be enabled by default because the OSD hardware is on a daughter-board which is sold separately.

Enabling by default, in code causes these issues:
1) code size bloat , no need to call `intFeatureSet`
2) confusion between `DEFAULT_FEATURES` in target.h and reality.
3) incorrect and wrong boot sequence and wasted CPU cycled for SPRacingF3NEO users that do not have an OSD daughter board connected.

To fix:
a) Remove the call to `intFeatureSet` from here: https://github.com/betaflight/betaflight/blob/master/src/main/fc/config.c#L597

b) Update targets and set DEFAULT_FEATURES so that FEATURE_OSD is included as appropriate.  The SPRacingF3NEO target should NOT have FEATURE_OSD on by default.

blckmn, 2017-02-19:
https://github.com/betaflight/betaflight/pull/2439 will resolve this for you.

martinbudden, 2017-06-04:
Addressed by PR https://github.com/betaflight/betaflight/pull/2439

closed 2017-06-04 by martinbudden