betaflight/betaflight #2442: GPS UBLOX7 Betaflight not receiving data

red-barran, 2017-02-20:
Using an NZ GPS module on UBLOX 7 protocol Betaflight 3.1.5 does not display data in configurator or OSD. Enabling serial pass through for the UART displays GPS data in the CLI, but the configurator does not recognise that a GPS receiver is present. GPS receiver confirmed working through U-centre via the flight controller using serial pass through.

mikeller, 2017-02-20:
What flight controller are you using? GPS support has been disabled on all but F4 based boards, due to limitations in the available flash memory.

red-barran, 2017-02-20:
It is the new Betaflight f3. Apologies if I missed that caveat about f3 versus f4 boards. The Betaflight F3 board is advertised to "work with all the features of Betaflight"

ldonnay, 2017-02-20:
Hello,
I am having the same issue on an Omnibus F3: GPS can be assigned in Port tab, and active in Configuration tab, but it is blocked on NMEA. I cannot switch it to UBlox. I tried all the ports, without success. I was also the case on 3.1.3 and 3.1.4. On 3.0.1, GPS was working fine, i.e. recognized.

mikeller, 2017-02-20:
@ldonnay: Read my comment above.

red-barran, 2017-02-21:
The opening statement of the FPVModel website description for the Betaflight F3 board is quoted:
"BetaflightF3 is the flight controller designed to use every possible feature of Betaflight.
I understand GPS isn't a priority for Betaflight, from my perspective GPS speed, altitude and position would be great. Position hold would be nice, and I'd assume that return to home is where the substantial additional code would be necessary

mikeller, 2017-02-21:
@red-barran: Yes. Unfortunately the reality has caught up with this statement, and the flash space that is available on the MCU is not something that can be changed in an easy way, once the type and package of the MCU, and the design of the board around it has been decided on. Betaflight focuses on providing optimal flight performance, and as a part of this, the way it is built has been changed from optimising for minimal space requirements to optimising for best performance and most constant loop times. This required for some features to be removed for F3 based boards.

If you want the features that you describe above, then you got a number of options:
- look into using iNav. This project is focusing on advanced navigation;
- build your own custom build of Betaflight, with only the features you want enabled. It is not that hard, and we can provide you support for it.

red-barran, 2017-02-21:
Are there any plans to release a Betaflight FC running F4?

Alternatively might there be a possibility of "fitting in" the code to support basic GPS speed, altitude and position coordinates?

It would probably be viewed by the community as beneficial to include some note in Betaflight that GPS won't work on F3 boards as the presence of multiple references to GPS does create the perception that it is a supported feature and it would be preferable to minimise unnecessary time spent troubleshooting.

ldonnay, 2017-02-21:
@mikeller I saw your comment. Like red-barran, Speed and altitude information are nice features. If there is a way to get this back on F3 boards, on slower frequencies, that'd be awesome.
Thanks

mikeller, 2017-02-21:
@red-barran: Re Betaflight F4 FC you'd have to ask @borisbstyle, the Betaflight F3 is his brainchild. ;-)

@ldonnay: We understand that, and we are certainly not taking the decision to remove features lightly. Whenever we have to do it, we try to make the choices in line with what serves the community best.

There are ideas being thrown around of moving towards a modular firmware, or an online build system (like the one that OpenTX uses) that would allow the users to customise their firmware with the features they want.

Ede2016, 2017-03-15:
I guess this is only partly related to this thread.
Does the S Port FrSky GPS Sensor senor still work with telemetry with an F3 board?
I like to use the newest BF and I would like to use GPS with FrSky Telemetry to be able to find my quad according to the GPS coordinates sent per telemetry to my Taranis - even if this data is not used in BF anymore.
Does anybody know if this works?
Thanks

mikeller, 2017-03-15:
@Ede2016: It should still work, SmartPort allows for more than one device to be connected to the RX concurrently. Just give it a go.

Ede2016, 2017-03-15:
@mikeller: Thanks for the info. I have to buy the GPS Senor first. This is why I asked because I don't want to buy it only to find out it does not work.

red-barran, 2017-03-15:
Yes, wouldn't it be irritating if you went and bought a GPS sensor because all the documentation and vendor advertising stated GPS was supported, you attach it to the Quadcopter, solder it in, spend hours diagnosing why it won't work, then find it was silently dropped from the code

mikeller, 2017-03-16:
@red-barran: The point is, if the sensor sends the data directly to the RX via SmartPort, there is no support needed in Betaflight. The question that remains is if the RX / TX can handle the data from Betaflight and another sensor being sent concurrently over SmartPort. And without having access to test equipment, answering this will be difficult, since the FrSky code is not open source.

red-barran, 2017-03-16:
That's not my point

mikeller, 2017-03-16:
Then your point seems to be different from what the author of this issue was asking for. Maybe you should go and ask the vendor of your hardware to release firmware that lives up to their promises.

red-barran, 2017-03-16:
Am I not the original author of this issue?

mikeller, 2017-03-16:
Sorry, my bad, I only read back to @Ede2016's post.

Ede2016, 2017-03-16:
@red-barran: I understand your frustration and you made your point. And the BF developers explained the situation. They develop the software for free and it's up to them what they do. We have no right to expect that everything works in the way it worked before - even if this might be frustrating.
Your issue is clearly about the promise of the BFF3 vendor. Maybe contact them and ask for your money back but don't blame the BF developers who spend a lot of their time to make a very good product. I think you should respect that.

red-barran, 2017-03-16:
Ede2016, I'm glad I was able to help you avoid unnecessary financial and time expenditure. You're welcome. Don't mention it.

closed 2017-03-16 by red-barran

mastrogippo, 2017-07-18:
Can you guys make it EXTREMELY CLEAR in the documentation, or even in the GUI with all the red banners, maybe in the GPS tab? I spent all evening debugging it and only now found this. I'll go on recompiling from source :)

jflyper, 2017-07-18:
@mastrogippo It's not that easy as GPS support availability depends on targets and firmware revisions...