betaflight/betaflight #2447: CC3D Evo Problem

astrobond, 2017-02-20:
Hi all :)

I moved yesterday from cleanflight to betaflight 3.1.5 on my "old" CC3D Evo (CC3D with spektrum sat port), and after a lot of tests, reflash 3.1.0, and back 3.1.5 there are two problems i don't find solutions :
1) My Spectrum Satellite SPM4648 is always in "binding mode" when i restart the CC3D Evo, even with tx on, i have to rebind each time i want to fly.
2) Impossible to configure the Leds, the LED Strip Panel is indicating "remaining leds 0 (zero)" and i can't add any leds, on the screen. (tried to add/remove LED_STRIP in configuration panel, reboot, clear leds, etc, still zero led remaining message).

Any ideas ?

Thank's for reading, Best Regards, astrobond.

mikeller, 2017-02-21:
2. is because support for LED_STRIP had to be removed from CC3D for 3.1.5 in order to make it fit into the very limited flash space available. Sorry.

astrobond, 2017-02-21:
@mikeller Ok thank's for the info :)

astrobond, 2017-02-21:
Last news: i juste flashed BF 3.0.1, all is ok with my CC3D evo setup, LED_STRIP ok, Spek Sat keep binded :)
(and micro minimosd on mainport ok)

alexandreBoussemart, 2017-03-01:
Bonjour,

J'ai flashé ma cc3d evo en betaflight 3.1.6. Tout marche correctement sauf le satellite. Je l'ai branché sur le flexport et dès que j'allume la carte il passe toujours en mode bind.
Pourtant j'ai bien mis le uart3 en serial_rx
Par ailleurs, est-il possible de monter un buzzer hobbyking sur la evo ?

j'ai besoin d'aide
merci
alexandre

astrobond, 2017-03-02:
Bonjour alexandre J

 

Meme probleme d’autobind en 3.1.X ici, j’ai du repasser la CC3D en BF 3.0.1 pour que tout fonctionne normalement.

 

Jamais testé le montage buzzer sur CC3D, mais logiquement sur le signal moteur 6 on recupere de quoi faire marcher un buzzer, mais il faut faire un montage spécifique CC3D avec un transistor et une resistance en plus d’un buzzer « standard » pour carte type naze32 ou SPF3 par exemple.

 

Hope this help ;)

 

Etienne.

closed 2017-03-15 by AndersHoglund

earthzara, 2017-10-22:
How you wire  led strip   to cc3d evo ? 
you can Guide me?