betaflight/betaflight #2449: KIWI F4 - Enabling LED_STRIP deletes all settings in EEPROM (3.1.5)

micmax79, 2017-02-20:
I recently bought a Matek Tail Light WS2812B including buzzer. The tail light is connected according to the manual: B- to Buzzer- pin on FC; 5V/GND is connected to corresponding supply pins on FC; DIN is connected to DATA output on bottom side of FC (as mentioned in WIKI).

The buzzer can be controlled via RX without any problems. The LEDs, once activated with the LED_STRIP option, light up white. As soon as I try to address/configure each single LED and press SAVE, the LEDs turn off and the complete EEPROM content is deleted (this failure does not always occur; I suppose it is dependent on specific settings for each LED; needs more systematic testing).

A similar problem regarding the CC3D is reported in #2250 (in my case, I only have to reconfigure the flight controller).

mikeller, 2017-02-21:
Hmm. Can you try repeating the same procedure, but this time with the LED strip disconnected? One suspicion I have is that the LEDs might overload the power supply, causing errorneous behaviour during the EEPROM write.

blckmn, 2017-02-21:
I agree with @mikeller, definitely try without the LEDs powered up.

micmax79, 2017-02-24:
Unfortunately, I'm not able to reproduce the failure. As proposed I removed the DATA output from the light bar, configured all LEDs in Betaflight, saved and then soldered the DATA output back to the tail light. The LEDs lit up as configured. I did some more tests (e.g. enabled all LEDs, buzzer, connected my receiver) in order to increase the load on the 5V rail. But as I said, the failure could not be reproduced.

Maybe the manual of the tail light is missing some information; it says: "If DATA is connected to FC, Pls disconnect it & re-energize the board when changing FC mode to TX mode." Maybe you also have to disconnect DATA if you switch from TX to FC mode...

closed 2017-02-24 by micmax79

mikeller, 2017-02-26:
Ha, I only now read the product description for this tail light. Turns out it contains other active electronics besides just the WS2811 compatible LEDs, and is capable of driving the LEDs from other inputs besides a WS2811 format signal. This makes it likely that the onboard electronics on the tail light being in the wrong configuration was the reason to blame for your problems.