betaflight/betaflight #2452: Add PID Profile Switching

Eganwp, 2017-02-21:
Keep meaning to log this, but we need to add a way to switch PID Profiles via the Adjustments tab in BFC. (I know a separate PR will have to be logged in BFC as well). As it stands now, you can do everything you want via the Adjustments tab in BFC, including "Rate Profile Selection" as a drop-down option, but there is no option for "PID Profile Selection" for some reason. The old way CF stick commands would work, but most people I've asked have no idea those even exist, so there's no reason we shouldn't add this feature as the "Rate Profile Selection" feature is already there and implemented, so very similar code.

mikeller, 2017-02-21:
Thanks for your input. 'live' PID profile switching (and with it probably also mergeing of rateprofiles into the PID profiles) is already on the roadmap. The reason it hasn't been done is that it's a bit more complicated than this, since the current implementation of PID profile switching essentially does a reboot in order to load the new profile's values. This is not needed any more, but it means it will take some effort and testing to make sure profiles can be switched in-flight without a reset and crash.

Eganwp, 2017-02-21:
Good input, that makes sense for sure, thanks Mike. Reason I'm asking is it all really boils down to the fact that I (and I'm sure others) would really would love to change a 3 position switch on my Taranis to switch PID profiles (one for racing, one for acro, one with test PIDs) just as I do rate profiles now. :) Glad it's on the roadmap, I didn't realize it was. 

- Travis

mikeller, 2017-02-21:
But you are aware that there is stick commands that can be used to switch profiles (when disarmed obviously ;-))? This way, you don't even have to use a switch / channel.

Eganwp, 2017-02-21:
Ya, that's what I meant when below when I mentioned the "old CF way" of stick command profile switching. Thanks again Mike.

Dave50W, 2017-11-09:
We should bring back stick commands it was so easy. 

Or We should have it both ways as I haven’t got a spare channel to put my profiles on a switch, 

I can’t find where it explains how to set up profiles the new way so I’m stuck without either

mikeller, 2017-11-09:
@Dave50W: Profiles are the same as they always were, except that the rateprofiles are no longer part of the profiles, but independent. This means stick switching of profiles works just like it always worked.

Dave50W, 2017-11-09:
@mikeller your right I run the same pids in my two profiles and only use it for changing rates, can I only change rates with a switch now? if yes how is it done thank you.

dave

mikeller, 2017-11-09:
@Dave50W: You can switch between 3 rate profiles with the 'Adjustment Range' facility. We are looking into the best way extend this to more than 3 rate profiles at the moment, and will have a solution in time for 3.3.

richard-scott, 2018-01-13:
I would really like to try and change to a different PID profile via an AUX channel setup in the Adjustments Tab but I don't see a "Profile Selection" option in the drop down. I'm assuming that's how this will work when implemented?

Is it still the plan to include this in 3.3?

Dave50W, 2018-01-15:
Changing with Aux would be good I do prefer the stick commands which leaves the aux’s for buzzer, turtle, lights & arming.

mikeller, 2018-01-15:
The problem is that switching profiles currently requires a reboot of the flight controller for the new values to be activated. Doing this through the adjustments tab would be awkward, if not dangerous (if done in flight).

There is probably not much work left for this requirement for a reboot to be eliminated, but it's still not as easy as just adding it to the RC adjustments, and it will require extensive testing to make sure all of the settings are applied after the change.

Dave50W, 2018-01-15:
I wouldn’t think you’d need to adjust mid flight I was able to quickly use stick commands and that worked great I now use betaflight or osd to change which is a hassle can we not roll back to stick commands leaving our aux channels for all the other things we use?

mikeller, 2018-01-15:
@Dave50W: It's not so much that you'd need to change PID profiles mid flight - but with the current implementation if you did so accidentally, the ensuing reboot (and obviously disarm) would surely cause you to crash. Stick commands for profile switching are still supported.

Dave50W, 2018-01-15:
Yes pids are supported sorry I mean the rc rates to switch from race to freestyle rates used to be stick commands with pids

mikeller, 2018-01-15:
@Dave50W: There is work under way to identify how this can best be done.

Dave50W, 2018-01-15:
That’s great mate tankyou @mikeller

marko-biedermann, 2018-07-13:
Has the PID profile life-switching  been implemented yet? Like to use it on my tilt rotor quad plane, where I want to disable both rates and PID when switching to airplane mode.

mikeller, 2018-07-13:
@marko-biedermann: No.

mccolld1, 2018-10-29:
Has any progress been made on this request (pid profile changes via tx stick)?  
If not, is it still a work in progress?

mikeller, 2018-10-29:
@mccolld1: Nope, still stuck on identifying a solution to https://github.com/betaflight/betaflight/issues/2452#issuecomment-357602376.

JustAskDave, 2019-04-11:
Hey, I notice this is still not implemented, cool, but I was hoping to add the max angle into my setup but its locked to the Profile and not the Rate Profile.  Is there anyway this could be attached ot the Rate Profile instead?   I am setting them up for school kids and wanted a 3 pos switch for beginner, medium and advanced.  Currently I am changing the RC Rate, Expo and Motor Max speed on the switch. Throwing the max angle in there too would just make it perfect :)

henrytseng, 2019-07-03:
I agree.  This would be a great feature.  It would be very useful to change PID profiles.  Would love to see this happen.

Eganwp, 2019-09-01:
Just wanted to follow-up on my request to see if any progress has been made to support PID profile switching in flight as I mentioned in OP. 

Reason I would love this feature is because I run different D_min settings, different options turned on for freestyle and different setpoint transitions on my freestyle vs race quads and it’s nice to fly both styles with all types of quads just by flicking a switch. :)

Thanks!
- Travis

mikeller, 2019-09-01:
@Eganwp: See https://github.com/betaflight/betaflight/issues/2452#issuecomment-433830107.

timmyknuckles, 2019-09-17:
> I agree. This would be a great feature. It would be very useful to change PID profiles. Would love to see this happen.
> 
> Would it be possible to allow change while disarmed?

I'm also wondering this.

Pavel-DV, 2019-10-07:
@mikeller Can this function be made so that the profile changes upon reboot?
For example, like this: The transmitter switch is set to the "1" position, the battery is disconnected from the quadrocopter and reconnected, and it works in the "1" profile. If you switch the switch to the "2" position and turn the battery off and on again, the "2" profile will turn on, and so on. Is it hard to implement? It would be very helpful. Please do.

mikeller, 2019-10-08:
@Pavel-DV: This could probably be done - have a 'special' adjustment range that is only evaluated once, on booting, after the RX connection has been established. But then at the same time, this seems to be super redundant, as you can simply use stick commands to switch PID profiles in this case.

Pavel-DV, 2019-10-10:
@mikeller Thanks.
I just did not know that there are stick commands.

oscar-yankee, 2020-11-25:
Hello, just following up on this request. Would like to be able to switch profiles via aux switch as opposed to OSD. I have different angle limits set on my mini whoop for indoor flying and would be  nice to "gear up/down" on the fly. Any progress?

mikeller, 2020-11-25:
@oscar-yankee: See https://github.com/betaflight/betaflight/pull/10329#issuecomment-723599071 - there still is a rather long list of settings that will need to be tested for compatibility with in-flight switching, and potentially adapted to support it.

robertrypula, 2020-12-19:
Looks like I want to achieve similar thing as @oscar-yankee. I came into this thread while trying to setup two modes on my 75 mm drone. I want to set two modes - both for **Angle** flying:
1) **Fast mode for me**
    - _PID Profile Settings_
      - Angle Limit: 45 degrees
    - _Rateprofile Settings_
      - Throttle limit: OFF
      - Throttle Limit %: 100
2) **Slow mode for my kids**
    - _PID Profile Settings_
      - Angle Limit: 10 degrees
    - _Rateprofile Settings_
      - Throttle limit: CLIP
      - Throttle Limit %: 40

There is no problem with Rateprofile inflight adjustment but as far as I read here there is big problem with changing PID profile without rebooting the craft. Maybe it would be possible to add **inflight adjustment** only for **Angle Limit** value? Actually that's the only option from the PID Profile Settings that I want in order to switch between **super fast** when I fly and **super slow** when my kids fly.

[image]

JustAskDave, 2020-12-19:
> Looks like I want to achieve similar thing as @oscar-yankee. I came into this thread while trying to setup two modes on my 75 mm drone. I want to set two modes - both for **Angle** flying:
>
> 1. [...]

That's exactly what I asked for, about a year ago, for very similar reason. We have a kit that school kids build and currently have a switch for throttle range and rates, then the stick commands for angle limit. Would love to have the angle limit under the rate profile switch.

robertrypula, 2020-12-19:
Could you elaborate more on "stick commands for angle limit"? I'm new to the hobby so probably I'm missing some ultra obvious thing ;)

JustAskDave, 2020-12-19:
> Could you elaborate more on "stick commands for angle limit"? I'm new to the hobby so probably I'm missing some ultra obvious thing ;)

Haha, no probs! The rateprofiles you can easily set to a switch, but the profiles (where you set the angle limit) you can select by the old stick commands. Throttle down and left, roll left for profile 1, pitch forward for profile 2, roll right for profile 3.

https://images.app.goo.gl/KcM7UFuGpSxEqRr19

robertrypula, 2020-12-19:
OMG, you just made my day (and my kids day) :D Of course it's a workarround and still it would be nice to change PID profile via AUX in flight but for now I'm super satisfied with your solution! Thanks a lot man! 👍

frizzle101101, 2021-06-13:
I would like to chime in and agree that being able to change the pid profile via switch simply like the rate profile would be very usefull an i agree about an option for allowing for inflight or not off by default kind of thing. 

Im not sure but to me this may interfere with the battery selectable pid profile feature, it may be necessary to make battery pid profiles as base profiles selectable by the turnon of the quad.  Then just have pid profiles which can be changed to with aux channels, always having the first or default profile the one that boots from the battery profiles

DzikuVx, 2022-05-30:
This is handled by Logic Conditions now, so profiles can be switched in flight

hickscorp, 2023-01-18:
Bump, this would be really useful mid-flight when you realize you took off outdoors with the indoor PIDs selected and vice versa :)

miketanct, 2023-05-22:
Yes, would be very useful if able to switch flight characteristics mid-air in certain types of drone flying, such as Drone Soccer. https://www.youtube.com/watch?v=JA7flsBpl8Q
https://fidachampionship.com/

SuicideQuad, 2024-06-09:
Super interested in this feature!

haslinghuis, 2024-06-09:
Needs to address https://github.com/betaflight/betaflight/issues/2452#issuecomment-433830107 but simply could be done using 

[image]

SuicideQuad, 2024-06-15:
> Needs to address [#2452 (comment)](https://github.com/betaflight/betaflight/issues/2452#issuecomment-433830107) but simply could be done using
> 
> ![image](https://private-user-images.githubusercontent.com/8344830/337995461-979b093c-1e1f-471d-a033-e59c337c1ddc.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTg0NTkyNTgsIm5iZiI6MTcxODQ1ODk1OCwicGF0aCI6Ii84MzQ0ODMwLzMzNzk5NTQ2MS05NzliMDkzYy0xZTFmLTQ3MWQtYTAzMy1lNTljMzM3YzFkZGMucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDYxNSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA2MTVUMTM0MjM4WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9ODg3NGQzMDYwZjFmOGRlMTc1ZTU2Nzg2MmI4OWZkZWYwNGMyYTQyNzc2MDlkZjVjZjY0Yzg1M2ZjOWQ3Mjc0MSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.gIljnImN9RSOxLpJbTYRIhOq3oOCgZNxK-hqqxRoYvU)

this does not work in flight, correct?

vedmant, 2024-09-06:
+1, would be really nice to have this in-flight for simpler PID tunning so flight characteristics of different profiles can be compared in a single flight.

alexb1352, 2024-10-17:
Hey @devs
How is it going? It's been 7 years since it was planned to be implemented 
Anyways, what about allowing to assign it on a switch only when not armed? I personally would be happy just with that. Thanks

jumpstartfpv, 2025-03-26:
Can this be implemented now with the newer version of betaflight? I was looking forward to using this to compare PID tunes on a quad and use two profiles for two configurations—one with a GoPro and one without.

robertrypula, 2025-03-26:
It's probably lost in the noise of this thread but it's already possibie via stick commands:

[image]

miketanct, 2025-03-26:
It's also probably lost in the thread noise that the main reason for this request is to have the ability to make the change mid-flight, thus the need to have it selectable via aux channels rather than stick movements. 🙂✌️

JustAskDave, 2025-03-27:
Any update on this option?  I am really just wanting the Angle Limit to be adjusted by switch, so throwing that into the PID Profile to Rate Profile would be perfect...   

I have a small drone kit we build in Schools, and I use Angle Limit 15°, 30° and 55° for beginner, intermediate and advanced to reduce the speeds and sensitivity, whilst I have three Rate Profiles to go from low rates and 50% throttle, medium rates and 75% throttle, and high rates with 100% throttle.  Would love to activate both simulataneously by switch.  I originally raised this 6 years ago before I saw this thread and my request was merged into this one.  But alas 7 years on from the original request and still no change!

Should it just be closed off ?

JustAskDave, 2025-04-01:
Uodate to my comment...  Just programmed a little iFlight 25 pusher in my original scenario on 4.5.1 and now the entry for Angle Limit on the Profile tabs is being implemented / added in the Rate Profiles.  I have Profiles 1, 2, & 3 with 15, 35 and 65 degrees. If I leave set to Profile 1 but change rate profiles with a switch the drone will tilt to around 35 degrees. If I leave on Profile one and select Rate Profile 3 it tilts to around 60 degrees.   If I select Profile 2 and Rate Profile 3  it goes to just past 90°. If I select Profile 3 and Rate Priofile 3 it will flip right over (120 degrees?? then continues to level again).   Has someone been fiddling!!

JustAskDave, 2025-07-16:
> Can you share how you did it? Thanks

It was a bug in 4.5.1.  Now updated to 4.5.2 and its back to normal operations.  Not sure what it was doing but it was replicatable every time on that version.  I was hopeful there for a minute!   But seems if it can be done by a programming bug it should be able to be done intentionally...

height888, 2025-07-16:
> It was a bug in 4.5.1. Now updated to 4.5.2 and its back to normal operations. Not sure what it was doing but it was replicatable every time on that version. I was hopeful there for a minute! But [...]

Tested 4.5.1 version on my BATAFPVF411, setting  Profiles 1, 2, & 3 with 15, 35 and 55 degrees,change rate profiles in flight，nothing has changed in angle limit .

miketanct, 2026-10-04:
It's near the end of 2026. Can we please have the ability to toggle the Angle Limit between two or three different values while the drone is flying?