betaflight/betaflight #2457: Disable TBS SmartAudio / IRC Tramp via AUX channel

19Gerhard85, 2017-02-21:
Hey guys,

is it possible to add functionality to disable the TBS SmartAudio / IRC Tramp feature via AUX channel?
If so, you could use the mic when you don't need to change VTX settings.

Greets

jflyper, 2017-02-22:
@19Gerhard85

Tramp has dedicated audio wire, so it wouldn't be a problem.

SmartAudio driver, when armed, stops talking to the device.

Do you want something else?

19Gerhard85, 2017-02-22:
@jflyper 
Thanks for you response.
Do you know if the Unify has a problem with when the mic and the SmartAudio is connected at the same time?

jflyper, 2017-02-22:
Never tried it.

Should have no harmful effect; it's only 1vpp anyways. I've also measured dc resistance of the smartaudio wire and was something like 13Kohm or something. Nothing special.

19Gerhard85, 2017-02-23:
Thanks for the information.

closed 2017-02-23 by 19Gerhard85