betaflight/betaflight #2459: SP Racing F3 Deluxe & FS-X6B Receiver - AETR consistently and randomly dropping to failsafe

RyanSA, 2017-02-22:
Running the above setup on BF 3.1.5 as well as flashing back to 3.1.

In BF Receiver tab, AETR commands are picked up, but every couple of seconds all channels just revert back to failsafe value. This happens consistently.

I'm new to this but happy to provide a CLI dump with regards to the above behavior.

Thanks!

mikeller, 2017-02-22:
How are your RX and flight controller connected?

This behaviour is most often caused by one or more of the channels being detected as a an invalid value, as determined by being outside of the range of `rx_min_usec` to `rx_max_usec`. Adjust these values to work with your RX.

RyanSA, 2017-02-22:
Signal wire connected to RX on UART 3. GND and 5V connected to PDB.

rx_min_usec is by default 885, so should I just keep upping the value until the channels don't drop anymore?

mikeller, 2017-02-22:
***Lowering*** `rx_min_usec`. And increasing `rx_max_usec`.

RyanSA, 2017-02-22:
Thanks, will give it a try this evening and report back.

RyanSA, 2017-02-23:
How do I lower rx_min_usec and increase rx_max_usec in Betaflight as the CLI won't accept these commands.

mikeller, 2017-02-23:
` set rx_min_usec = X` and `set rx_max_usec = Y` should work in the CLI.

RyanSA, 2017-02-23:
Thanks, wasn't inputting the set command. Will report back.

RyanSA, 2017-02-24:
I've adjusted the `rx_min_usec` and `rx_max_usec` values and all that happens is that throttle randomly just drops to whatever `rx_min_usec` is set to and all other channels revert to 1500. I've even disabled failsafe on Betaflight completely but the same behavior persists.

mikeller, 2017-02-26:
This is exactly the behaviour that it is meant to show whenever there is no valid RX input signal detected. Could it be that your RX, the connection RX - flight controller, or the power supply for your RX are unstable (or even your TX or the connection TX - RX)?

Noctaro, 2017-02-26:
I dont know if it belongs to the same bug. But on CC3D i had to remap AETR to some very exotic mapping to get it back to work. All channels were mixed up. Back to 3.01 everything works as intended AETR mapping works as it should. Further i was not able to fly it after correcting the channel mappings in CLI. (Took me quite some time to figure out the right mapping.) As soon as i armed it, on hands everything worked normal. At takeoff after a secound or so it went in some dangerous angle and hit the wall. Did no further testing on this setup, but i will do so if someone has an idea.