betaflight/betaflight #2460: stable flight mode, more and more offset

qdog2012, 2017-02-22:
when i am using angle mode, becoming more and more offset, but in cleanflight that's not gonna happen like this.

mikeller, 2017-02-22:
Your accelerometer is not calibrated. Put your craft on a level surface, and click 'Calibrate Accelerometer' in the configurator 'Setup' tab.

mikeller, 2017-02-22:
It looks like you are misunderstanding the instructions `serialpassthrough` sets up a passthrough ***from USB*** (VCP or otherwise) ***to the UART you choose***.

qdog2012, 2017-02-23:
I did. After flying, at the beginning of the time is good, after a while began to shift, and the offset angle is more and more big.

mikeller, 2017-02-23:
Could be caused by vibrations, exacerbated in Betaflight by the higher motor update rates. See #797 and try soft mounting your flight controller.

fftunes, 2017-02-25:
I still don't understand the logic behind why acc is even allowed to "re-orientate" itself. The acc is calibrated, and the result is saved for a reason, and this could be re-read any time. No?