betaflight/betaflight #2461: how to passthrough any uart to vcp?

andre92, 2017-02-22:
I couldn't find this in the wiki. I am using an SP racing F3 evo with VCP, smartport on uart 1, serial rx on uart 2 and minimosd on uart 3. I want to configure serialpassthrough from vcp to uart 3 but can't seem to get it working. Any ideas?

mikeller, 2017-02-22:
Yes. Read https://github.com/cleanflight/cleanflight/blob/master/docs/Serial.md

andre92, 2017-02-22:
I read that, but it appears it passes through only any other uart to uart 1 and not the vcp

mikeller, 2017-02-22:
It looks like you are misunderstanding the instructions `serialpassthrough` sets up a passthrough ***from USB*** (VCP or otherwise) ***to the UART you choose***.

andre92, 2017-02-22:
Still not working for me. I type "serialpassthrough 2 115200", that should connect uart 3 to the vcp. Battery was plugged in to keep the osd powered. I ended up using ftdi to configure my osd.

mikeller, 2017-02-22:
Weird. Works just fine for me here. What is the response you are getting after `serialpassthrough 2 115200`?

andre92, 2017-02-22:
actually there is no response until I cycle it, and port appears in use, even after I disconnect and close configurator. Do you think my FC could be freezing?

closed 2017-02-22 by andre92