betaflight/betaflight #2471: Can't set SBUS/automatic reset to Spektrum1024 -SP Racing F3 + X4R

Skpinus, 2017-02-22:
I've just flashed my SP Racing F3 FC from betaflight 3.0.0 to 3.1.5 and now suddenly I can't set "serial receiver provider" to SBUS. Everytime when I select SBUS and do a "save and reboot", it resets to SPEKTRUM1024.
I've double checked everything, I'v got UART3 (where my x4r is connected) set to serial RX (everything else left to default), receiver mode set to "serial-based receiver".
I've tried maunaly set it to SBUS via CLI, it did nothing.

Funny thing, it worked without any issue on BF 3.0.0 and I didn't changed anything, just flashed the FC.  I've tried to reflash the FC with 3.1.5 again, that also did not reseolve anything.

Any ideas?

chrigu70, 2017-02-22:
Sorry for no solution but i have the very same problem with my brandnew Motolab Tempest ver.1.3 and Frsky X4R-SB that i want to set up now (both are brandnew).
Everything appears just as you described it. 
I dont know if it would have worked with factory flashed BF 3.0.1 because i directly updated to 3.1.5.
I select SBUS then save and reboot and it just resets back to SPECTRUM1024.
Therefor the receiver tap shows extreme erratic behaviour.

Any suggestions?

AndersHoglund, 2017-02-22:
Can not reproduce on my SPRACINGF3. Selecting SBUS works just fine with 3.1.5. 
Try "Reset Settings" and start over.

chrigu70, 2017-02-22:
Just tried all versions down to 3.0.1 - and this one works fine.
Didnt try Reset Settings yet...

chrigu70, 2017-02-22:
Reset Settings didnt help.
3.0.1 works fine - at least on my Motolab Tempest.
Ill stick on that one for the moment.
Eager to see a solution to this problem.

AndersHoglund, 2017-02-22:
OK, Confirmed with BetaFlight Configurator 1.9.2, works OK with previous version 1.9.1
Not sure how to downgrade, other than down loading and installing that version from github.
https://github.com/betaflight/betaflight-configurator/archive/v1.9.1.zip

Reinsettle, 2017-02-23:
I'm having the same problem with IBUS after flashing 3.1.5. I reflashed to 3.1.0 that i was using and it is still switching it from IBUS to Spektrum1024. I Don't know what to do.

PietroSpina, 2017-02-23:
I had the same problem. It started after the most recent configurator update. (1.9.2) 
My work around was to use the CLI I did the following and it stayed.
`set serialrx_provider = SBUS`

Skpinus, 2017-02-23:
> AndersHoglund commented 8 hours ago
OK, Confirmed with BetaFlight Configurator 1.9.2, works OK with previous version 1.9.1
Not sure how to downgrade, other than down loading and installing that version from github.
https://github.com/betaflight/betaflight-configurator/archive/v1.9.1.zip

> 

That may be the reason behind the issue....I did update my configurator to the latest version just before flashing the FC to 3.1.5, I tottaly forgot about that.
When I get home, I'll try downgrading the configurator and reflashing again the FC to see if that solves anything. I'll report back later

closed 2017-02-23 by borisbstyle

TparXx, 2017-06-10:
Having the exact problem, did you ever find a solution?

Skpinus, 2017-06-10:
yes, update your Betaflight configurator - chorme app.  Version 1.9.2 vas giving me this problem, then I've updated the app to a newer version and it worked just fine

davkuzmin, 2017-10-14:
I'm getting this same exact issue on version 3.2.1 of the configuration. Going through the CLI didn't help either, every time the board reboots it defaults to the SPEKTRUM1024 provider.

jflyper, 2017-10-14:
@davkuzmin Type CLI "serial" and paste the result please.

davkuzmin, 2017-10-14:
@jflyper 

# serial
serial 20 1 115200 57600 0 115200
serial 0 0 115200 57600 0 115200
serial 1 0 115200 57600 0 115200
serial 2 0 115200 57600 0 115200

#

jflyper, 2017-10-14:
@davkuzmin You don't have any UART with Serial RX defined.

davkuzmin, 2017-10-14:
@jflyper Yes, I had them disabled at the moment I ran the command. I was following the [instructions this user suggested](http://www.dronetrest.com/t/aurora-90-rx-issue-betaflight-not-saving-sbus/5123/2). If edited my previous comment to explain that I tried enabling the different UART ports individually and together but the issue persists.

jflyper, 2017-10-14:
In CLI, do `defaults`.

When it comes up, paste these lines into CLI.
```
feature -RX_PPM
feature RX_SERIAL
serial 2 64 115200 57600 0 115200
set serialrx_provider = SBUS
save
```

davkuzmin, 2017-10-14:
That did it, but as soon as I power cycled the board, it defaults back to Serial-based receiver/SPEKTRUM1024

jflyper, 2017-10-14:
What is your FC?

jflyper, 2017-10-14:
Can I see the current `diff` output?

davkuzmin, 2017-10-14:
I'm using the [Realacc F3](https://www.banggood.com/Realacc-F3-V1_0-Betaflight-Brushed-Flight-Control-Board-Integrated-OSD-with-Buzzer-STM32-F303-MPU600-p-1121378.html?rmmds=myorder)

`diff` output is below

# diff

# version
# BetaFlight/OMNIBUS 3.0.1 Oct 18 2016 / 10:21:07 (48b7b4f)

# name

# mixer

# servo

# servo mix

# feature
feature -RX_PPM
feature RX_SERIAL

# beeper

# map

# serial
serial 2 64 115200 57600 0 115200

# led

# color

# mode_color

# aux

# adjrange

# rxrange

# rxfail

# master
set mid_rc = 0
set min_check = 0
set max_check = 0
set rc_interpolation = OFF
set rc_interpolation_interval = 0
set motor_pwm_protocol = BRUSHED
set serialrx_provider = SBUS
set airmode_activate_throttle = 0
set rx_min_usec = 0
set rx_max_usec = 0
set osd_rssi_pos = 3094
set osd_gps_sats_pos = 1410

# profile
profile 0

rateprofile 0

# rateprofile
rateprofile 0

#

jflyper, 2017-10-14:
Ah. Your firmware is fu**ing old, two generations old, and is not compatible with BFC3.2.x.

Please upgrade to 3.2.x and try again.

jflyper, 2017-10-14:
In fact, the serial RX provider (`serialrx_provider` variable) is correctly set as `SBUS`, but BFC is displaying it as `SPEK1024`.

davkuzmin, 2017-10-15:
@jflyper I updated my board's firmware to the latest OMNIBUS 3.2.0 and the problem was fixed. Many thanks!