betaflight/betaflight #2477: Using Dshot on a Nucleo F7 or F4 board

Plantgrower, 2017-02-24:
I am making a robot for PiWars 2017 and want to control four brushless motors using Dshot. Could someone please tell me what I need to make this work on a Nucleo F7 or F4 board. I see that Nucleo F7 is declared in Target but //#define USE_DSHOT is commented out in target.h . Is Dshot not yet working on this platform ?

jflyper, 2017-03-01:
Dshot on NUCLEOF7 is wip, I guess. It might work by just resurrecting the USE_DSHOT in target.c.

However, I don't see any point using Dshot or F7 for the platform nor BF for towards piwars; isn't there more appropriate nav-oriented ones?

Plantgrower, 2017-03-01:
I am only wanting to take out the Dshot part. I have  extracted the pwmWriteDigital function and I am working on the pwmDigitalMotorHardwareConfig. I just need a much simplified version of this function to work the four motors in Dshot 1200 mode with KISS 24a escs. It will be hard coded to the particular board I am using. I haven't used dma and pwm on STM32 before so it is rather a steep learning curve. If any one could give me advice on how to do this or any resources that would help that would be most appreciated.

Betaflight certainly compiles with the target Nucleof7 and use_Dshot uncommented.

mikeller, 2017-03-01:
@Plantgrower: But what is your rationale for wanting to use DShot? There are other protocols that are more common, and are supported by a wider range of projects / libraries than DShot. Using a new and 'exotic' protocol will not do you any favours, unless you can get something out of it that other protocols cannot give you.

Plantgrower, 2017-03-02:
I am using the motors as hub motors. The problem that Dshot helps with is that with other protocols it is difficult (impossible) to get them to turn slow enough. Dshot allows them to turn much slower and has more steps in the control. I am using magnetic rotary encoders with my motors and it makes the control much easier.

martinbudden, 2017-03-02:
@Plantgrower , have you tried low kV motors, eg gimbal motors which are available with 90kV, eg:
http://www.ebay.co.uk/itm/G-SUN-2208-Gimbal-Brushless-Motor-RC-Part-90KV-3mm-shaft-for-100-200g-SJ4000-E-/262382328495

Plantgrower, 2017-03-02:
@martinbudden I don't think they are powerful enough for what I want. The ones i have currently are Cobra 2213/950kv though they have some 700kv. Dshot 1200 at 32khz should allow better control.

martinbudden, 2017-03-03:
@Plantgrower , I was not suggesting you used those specific motors, rather they are just examples of low-kV motors. There are certainly more powerful low-kV motors available. Not also that if you run a high-kV motor at a very low speed, you will get a fraction of its normal output power.

Plantgrower, 2017-03-03:
@Plantgrower Thanks for the suggestion I'll have a further look. I'll need
a motor with standard size holes as I've already had the mounts machined in
aluminium. If I run a more powerful I can at least go faster when the need
arises. There is at least least one challenge where speed plays a part in
the result. For the moment I'm trying to work out how to get the compiler
stop complaining about the difference std::uint8_t and uint8_t ?

closed 2017-10-23 by jflyper