betaflight/betaflight #2484: DFU Mode Enables Buzzer on Omnibus F4 Pro V2

MiddleMan5, 2017-02-25:
Build Target: OMNIBUSF4SD
Method of replication:

1. Solder buzzer to dedicated Buzz+ and Buzz- pins (Mapped to B4 on Omnibus)
2. Enter BF Config Util, navigate to CLI, enter dfu
3. Board resets with buzzer stuck on

This issue is mildly annoying, I haven't yet been able to track it down as an addressed or configurable feature so I'm assuming it's a bug.

mikeller, 2017-02-25:
Question: Does this happen when you activate DFU mode by pushing the 'BOOT' button while plugging in USB as well?

In general, I am not sure if there is a fix for this. DFU mode is hardcoded, non-changeable into the MCU, and if it drives the B4 pin to low when being invoked, thusly activating the buzzer, then there will be nothing the firmware can do to stop this from happening.

jflyper, 2017-02-26:
@MiddleMan5 @mikeller I think the feature is built-in... :eek:

mikeller, 2017-02-26:
@jflyper: Did you find an explanation of why B4 is pulled down in DFU?

jflyper, 2017-02-26:
@mikeller The PB4 is designated as NJTRST (JTAG RESET) on reset and it's internally pulled up.

According to RM0090, reset state for PB4 is alternate function with PP + PU, which turns on the NPN driver transistor(s).

mikeller, 2017-02-26:
Ah, found it, thanks. Good to know that these pins are activated at every reboot (makes them a not - so - great choice for a motor out on a brushed micro. :-P)

closed 2017-02-26 by mikeller