betaflight/betaflight #2487: D Term Notch Filter issue.

titanrw, 2017-02-26:
Hi All.

I didn't know that notch_cutoff had to be lower than notch_hz.

It seems the GUI enforces this for gyro_notch filters.  For example, if I try and set a gyro_notch_cutoff higher than the gyro_notch_hz, upon reboot, the notch_hz has been zero'd out.

The GUI doesn't enforce this for d_notch settings.

For example, the correct setting: d_notch_hz=160 and d_notch_cutoff=100 (this is a 6 inch quad), works fine.

When I mistakenly set d_notch_hz=160 and d_notch_cutoff=220, the GUI accepts the numbers, but when the quad arms, I get motor 2 commanded to full throttle immediately.  Setting d_notch_cutoff=100, and it arms properly.  The only change was the filter setting.   I wouldn't have thought improper notch filters could cause the 'flip on takeoff', but it seems it can.

If the GUI zero'd out d_notch_hz with the same logic it did with gyro_notch_hz, then I assume this wouldn't happen.  As setting an improper notch_cutoff will simply end up with a disabled notch, instead of a motor at full throttle immediately at arm.

To be clear, to reproduce, set d_notch_hz=160, d_notch_cutoff=220.  REMOVE PROPS.  Arm quad.  Unexpected motor output ensues.  The motors work properly when commanded through the motors tab.  When arming and looking at the motors tab though, you can see their incorrect armed values.

fftunes, 2017-02-27:
If the GUI is protected against user error on gyro filters, same should be expected on dterm filters...

Good find. Though this is probably configurator issue, maybe better re-post there.

titanrw, 2017-02-27:
Thanks.  Will repost to the configurator page.

titanrw, 2017-02-27:
Reposted in configurator.  Ticket closed with response:

> These checks are currently only done in the firmware, and not in configurator. The check for the validity of the DTerm notch filter settings is indeed missing in the firmware.

It seems this is a BF issue after all?

PR #2515 (2017-02-28): Added validation for Dterm notch filter settings.
Fixes #2487.
closed 2017-02-28 by martinbudden