betaflight/betaflight #2491: Motors automatically start to spin as soon as quad is armed

trojanGoat, 2017-02-26:
My motor spin settings are "off"  and "airmode" is off by default.  As soon as I arm the quad, it will automatically spin the motors.  If I disarm wait 10 seconds and arm again, the motors spin faster and the quad starts to drift (possibly flip over if I wait longer)  I'm using Betaflight FC.

Here's my diff settings:

# Betaflight / BETAFLIGHTF3 3.1.5 Feb  7 2017 / 22:03:24 (4646f9d)
name Chimera

feature -BLACKBOX
feature -SDCARD
feature MOTOR_STOP
feature LED_STRIP
map TAER1234
serial 0 8192 115200 57600 0 115200
serial 2 4 115200 57600 0 115200
led 0 0,0::CW:1
led 1 1,0::CW:1
led 2 2,0::CW:1
led 3 3,0::CW:1
led 4 4,0::CW:1
led 5 5,0::CW:1
led 6 6,0::CW:1
led 7 7,0::CW:1
led 8 8,0::CW:1
led 9 9,0::CW:1
led 10 10,0::CW:1
led 11 11,0::CW:1
aux 0 27 0 1300 2100
aux 1 12 1 1650 2100
aux 2 25 0 1650 2100
set min_check = 1000
set min_throttle = 1040
set max_throttle = 1990
set digital_idle_percent =  3.000
set use_unsynced_pwm = ON
set motor_pwm_protocol = MULTISHOT
set motor_pwm_rate = 32000
set sbus_inversion = OFF
set ibat_scale = 235
set moron_threshold = 32
set deadband = 2
set yaw_deadband = 2
set acc_hardware = NONE
set pid_process_denom = 1
set osd_units = METRIC
set osd_cap_alarm = 1300
set osd_time_alarm = 3
set osd_vbat_pos = 2411
set osd_rssi_pos = 2070
set osd_flytimer_pos = 2423
set osd_ontimer_pos = 2455
set osd_flymode_pos = 2112
set osd_throttle_pos = 2272
set osd_vtx_channel_pos = 2048
set osd_horizon_pos = 200
set osd_current_pos = 2368
set osd_mah_drawn_pos = 2400
set osd_craft_name_pos = 2066
set osd_gps_sats_pos = 1410
set osd_power_pos = 2432
set vcd_video_system = 2
profile 0

set vbat_pid_gain = ON
set setpoint_relax_ratio = 88
set d_setpoint_weight = 250
set p_roll = 48
rateprofile 0
rateprofile 0

set rc_rate = 80
set rc_rate_yaw = 130
set thr_mid = 0
set thr_expo = 10

mikeller, 2017-02-26:
No surprise there: `set min_check = 1000` means that for any value above 1000, your throttle will be recognised as 'not at zero', which is probably just caused by normal jitter when your stick is at 0.

trojanGoat, 2017-02-26:
Hi, my min check is at 1000,  and the spinning i'm getting is not little twitches,  it almost seems like it's on airmode.

AndersHoglund, 2017-02-26:
If `min_check` is at 1000, what is your Tx throttle low endpoint?
Normally Tx throttle low endpoint is set 1000. You need some separation here, else you might get the effect mikeller described. Set your min_check at least 10 or 20 above Tx low endpoint. I.e. 1010 ... 1020, if Tx is at 1000.

trojanGoat, 2017-02-26:
Thanks guys,  setting min_check to 1020 for me worked!, closing ticket.

closed 2017-02-26 by trojanGoat