betaflight/betaflight #2492: Omnibus f4 OSD issues: cross hair, artificial horizon

Telnoi, 2017-02-26:
Certain enabled betaflight OSD features do not show up, such as:
-Artificial horizon
-Cross hair

jflyper, 2017-03-01:
They do show up in all available fonts.

Tested with v3.1.6 on
OMNIBUS F4 AIO (OMNIBUSF4 target)
OMNIBUS F4 V2 (OMNIBUSF4SD target)

Telnoi, 2017-03-01:
What could prevent these options from showing up?
This is my OSD with crosshair and artificial horizon enable.

https://youtu.be/cVtyCDWF1AE

jflyper, 2017-03-01:
Weird... There isn't even a faintest sign of them...

redneonrt, 2017-03-03:
Ive been having this same problem as well

jflyper, 2017-03-03:
@redneonrt 
> Can you checkout the cli variables `osd_horizon_pos` and `osd_crosshairs` ?

Telnoi, 2017-03-03:
Turned out it was an issue with one of the offered fonts. Changing it resolved the issue.

nathantsoi, 2017-03-04:
Which one?

jflyper, 2017-03-04:
@Telnoi 
I went through all the fonts and couldn't find the one that doesn't have glyphs for AHI and cross hair.

Telnoi, 2017-03-20:
Sorry, I don't remember which one. BF does not show which font is in use. I will try to reproduce this once a second omnibus arrives.

kifling, 2017-05-20:
I had this same problem, and changing fonts fixed it too. Bold worked, default and digitial did not. 

Not sure it matters, but my googles are the eachine vr-d2

beaverollie, 2017-07-09:
Agree with kifling...  osd was not working at all with my omnibus f4 fc and i reloaded each font until i found a few that worked in betaflight.  turns out its just a little glitchy sometimes.