betaflight/betaflight #2499: Stall Issues(?) DSHOT - Airmode ßF 3.1.5

Sherulez, 2017-02-26:
Hello,

I have encountered a problem since I flashed my flight controller to ßF 3.1.5.

### When I do an 'hard' roll my Quad keep spinning and crash.
This **only occur with** the **AIRMODE** feature.
The BlackBox file shows my motor 2 at 100%.

**Hardware information :**

- Quad : X (QAV-R RTF)
- FC : F4 Advanced Flight Controller - (MPU6000, STM32F405)
- ESC : Lumenier 30A BLHeli_S ESC OPTO (2-4s) DSHOT
- Motors : Lumenier RX2206-11 2350Kv

**Software information :**

- ßF v.3.1.5 (or 3.1.3, same problem now)
- BLHeli 16.6 (G-H-30)
- `motor_pwm_protocol = DSHOT600`

**BLACKBOX :**
[Blackbox_1](https://drive.google.com/open?id=0B0eBDY29hoSRQ1MxZjByZ0RFa28)
[Blackbox_2](https://drive.google.com/open?id=0B0eBDY29hoSRaUlPQ0hxQU90cTA)

**DUMP :**
[Dump_1](https://drive.google.com/open?id=0B0eBDY29hoSRWTY0a0d3emFHejg)

### **Already tried :**

- Change the `digital_idle_percent `
- Reflash ßF
- Reset everything to default
- Recalibrate the gyros and the accelerometer
- Reset the PIDs and Rates to the default settings
- Other frequencies for Gyros and PID loop
- Desolder the ESC and the motor 2 and invert it with the 3
- Remove the led strip (DMA?)
- Change the 'Startup Power' parameter in 'BLHeli Configurator' to 0.125

(Sorry for my English).

mjedlinski, 2017-02-26:
Hi. I have same issue with Kiss24 ESC, T-moror F40 and BetaflightF3.
I increase digital_idle_percent to 9, and it helps in normal weather conditions.
Today in very strong winds i have same thing, and I had to increase digital_idle_percent to 9.5.
It's to high than should be (when I arm on unstable ground copter try to slide (due to lift makes by the props)). Luckily my copter don't completely desync, it just ends flips and rolls in different, than expected, positions.
Maybe AIRMODE try to stop motor completely rather than slow it to digital_idle_percent ?

Sherulez, 2017-02-26:
Oh someone else in the same case as me.
I'm not really sure if that's good news ;-)
Thank you for your reply.

It is indeed the first solution that I tried but without success.

I have already tried to set the `digital_idle_percent` to `15`.
For details, I tried: `4.5`, `5`, `6`, `8`, `10` and `15`.
Tomorrow I will try to increase it to `20`.

Concerning the `MOTOR_STOP`, it is indeed disable in ßF.

As in the blackbox it is the commanded throttle value to the ESC. Not the actual motor speed _(ty waltr)_, the best solution will be that I film my quad in the third person, and that I slow down the video to see if the motor really stops. But I dare not risk a GoPro ;-)

I guess it's a failure of my FC or motors, but as I'm not sure I'm waiting to be sure before changing parts.

Sherulez, 2017-02-26:
## On RCGroups I found :

### **By BorisB 🥇 :**
[image]

### **By waltr :**
[image]

>1- min_throttle too low and ESC/motor stalls. Try increasing min_throttle.

**Done, nothing change.** `digital_idle_percent`

>2- Prop nut is not fully tightened and prop slips.

**Checked.**

>3- ESC or motor bad or weak. Replace parts if above doesn't fix issue.

**Done :**  
> Desolder the ESC and the motor 2 and invert it with the 3

borisbstyle, 2017-02-26:
If your motor shows 100% in blackbox and not being able to provide thrust  to correct the quad than the issue is really in your esc that has desynced most likely.
There is nothing that fc firmware can help avout that.

Sherulez, 2017-02-26:
Thank you so much for the time you give me ;-)
I understand that, but I did this:

> Desolder the ESC and the motor 2 and invert it with the 3.

So it's a problem with ESC 4 or 1 if I understand correctly ?

mjedlinski, 2017-02-26:
No. The problem is in motor or esc that shows 100% in BB.
If You change esc, maybe You should try to change motor.
IF I undestund correcrly in Your case is always the same motor? 
What happends when You do left roll or flip?

Sherulez, 2017-02-26:
> What happends when You do left roll or flip?

I have not tried yet.

No it is not always the same esc / motor. As I inverted the esc/motor 3 and the esc/motor 2. And the problem is still on the motor 2 in the blackbox.

I do not know if I am clear as I am french ;-)

Sherulez, 2017-02-26:
**About my props / motors  :**

I tried with HqProps and Lumeniers. I think the HqProps are lighter.

I flew about 16 times with this setup. I paid attention to the temperature of my motors during the first 3 flights: no problem. So maybe they have heated up during the other flights, I do not know (arf ..). (I did not have this dsync problem during those 16 flights).

I made some strong landing (bouncy bouncy) but nothing very violent.

I was running at 4.5% idle without a problem.

I have some magnets on my motors which are slightly brown on top (glue?).

**About my settings :**

Loops time : 8/2

As it is late in france, my to-do list for tomorrow morning:

- [ ] Try a left roll or a flip

- [ ] Break other props

- [ ] Try medium-high timing

- [ ] Changing demag to high

- [ ] Motors Temp. check

- [ ] Set startup power to 50% or 100%

Once this has been tried I will come back to you, **thank you** for your help.

PS : I have videos of my first flights, maybe that depending on the noise of the motors we could detect a problem ? From my little experience I hear nothing strange.

Sherulez, 2017-02-27:
Hello again,
So, I come back with some info:

- [x] Try a left roll or a flip

     Done, there is the BlackBox : [Left Roll](https://drive.google.com/open?id=0B0eBDY29hoSRMGw4eE05eG9BODA)
     **This time it is the motor 3 that seems to stall**

- [x] Break other props

     [There](https://cloud.githubusercontent.com/assets/25638207/23361955/a06c072c-fcf3-11e6-921a-5962cb90fe5a.jpg)

- [x] Try medium-high timing

     Nothing has changed

- [x] Changing demag to high

     Nothing has changed

- [x] Motors Temp. check

     Not hot at all

- [x] Set startup power to 50% or 100%

     Nothing has changed

So I have a desync on my esc / motor 2 and 3 if I understood correctly ?
If so, I really do not understand why it happened suddenly. My first 16 flights went very well.
So if I have to change parts, would that be my ESCs and motors?

mjedlinski, 2017-02-27:
Hi. Do You change rc_rate or setpoint weight?  Maybe You ask to much from Your hardware.
The motor desync because it must brake and suddenly go to full power in hundreds of milliseconds. If You roll right it will be right motor, if You roll left it will be left motor etc.

borisbstyle, 2017-02-28:
Do you have your ground wire soldered as well?

Sherulez, 2017-02-28:
As it is a rainy day, I can not make my flight trials.
:-(

Yes I check all the welds.

I have not tested in Oneshot, as soon as the weather is fine I will try.

I am even flashed BF 3.1.6 in doubt where my first flash would have gone bad.

I will try with defaut settings from this new installation.

Sherulez, 2017-03-01:
So I have :

- Flashed BF to 3.1.6
- (re)Flashed my ESC's to 16.6
- (re)Calibrated my ESC's
- (re)Selected DSHOT600
- (re)`digital_idle_percent` to `4.5`
- (re)Enabled AIRMODE

And ........ 
## Problem fixed !
**With default rates and PID's**

Well it's very weird because:

**1.** It worked for 16 flights with **my** settings(rates, pids, loop ...) **
2. It did not work with default settings after flashing in 3.1.5.
3. It did not work with  **my** settings after flashing in 3.1.5.
4. It works with the default settings of 3.1.6.

Now, I will, step by step, redefine my settings and see when the problem will come back.
(May be never ... Hope)

closed 2017-03-20 by Sherulez