betaflight/betaflight #2501: LED strip issues

hydra, 2017-02-26:
given the following config:

```
led 0 8,15:S:FW:2
led 1 9,15:S:FW:2
led 2 10,15:S:F:2
led 3 11,15:S:F:2
led 4 12,15:S:FI:0
led 5 13,15:S:FI:0
led 6 14,15:S:FI:0
led 7 15,15:S:AI:0
led 8 0,15:S:AI:0
led 9 1,15:S:FI:0
led 10 2,15:S:FI:0
led 11 3,15:S:FI:0
led 12 4,15:S:F:2
led 13 5,15:S:F:2
led 14 6,15:S:FW:2
led 15 7,15:S:FW:2
```

[image]

The following LED's misbehave.

LED 0 flashes red
LED 12 always displays the same color (very dark red)
LED 14 and LED 15 display larson scanner

The same config LEDs and config works fine in CF v1.x so the hardware and config is correct.

hydra, 2017-02-26:
additionally, the led init strip sequence that sets all LEDs to WHITE on boot works just fine, indicating the issue is a high-level issue rather than a driver issue.

mstrinzha, 2017-02-26:
Before fix:
https://www.youtube.com/watch?v=ZQUAQdIjTow

After fix:
https://www.youtube.com/watch?v=5preY0DYoVw

closed 2017-02-26 by mikeller

hydra, 2017-02-27:
great, thanks for the timely fix.

PR #2502 (2017-02-26): Fix Led Strip
This PR fixes #2501