betaflight/betaflight #2507: Quad flipping out in a hard yaw BF 3.1.5

allinarush, 2017-02-27:
Spracing f3, Emax Red Bottom RS2205, Storm 30A ESC, 4S.
I flashed to Betaflight 3.1.5 (latest release). The quad hovered and flew ok with default pids, but it just flipped out at random just hovering. I took it to the park to open it up and it did great except after a hard 180 yaw it flipped out again. I changed props, and it oscillated the entire time so I started adjusting the P values up and down. The oscillation did not stop until I reduced my RC rate from 1.0 to .65. 

So after flying around I started to try and open it up a little bit and again after a hard yaw 180 it flipped out again. I thought it was PIDS, but I went up and down on the P values for an hour and a half. To me I should be able to increase my RC rate beyond .65... that just seems low.
   
I had all of these components on a ZMR250 that I moved to a QAV-X 210. I was running cleanflight and it flew fine.

fftunes, 2017-02-27:
Sounds also like typical desync. Increase min_throttle for common motor protocols, or digital idle when using dshot.

fftunes, 2017-03-01:
You could also try to lower dterm setpoint weight to take some stress off the escs.
Or flash 3.1.6 which has lower defaults.