betaflight/betaflight #2508: BF 3.1.5 - SPRACINGF3EVO Frsky Sbus not working (on warlark)

oxivanisher, 2017-02-27:
I tried to use BF 3.1.5 (came with 3.0.1) on the [f3 warlark](http://www.overskyrc.com/warlark80-fpv-quadcopter-based-on-f3-or-f1-flight-control-board-p-753.html) with the frsky reciever on SBUS. I was unable to get it working and tried all the things that came to my mind including switching to PWM/PPM. No luck at all  getting any RCV inputs (the reciever was bound). If I switched to Spektrum SBUS, I had erratic inputs (which is correct since its a frsky RCV) and i knew that I am using the correct UART.

Then I switched back to BF 3.0.1, applied the same (frsky) configuration and it just worked. I can't say if they do something fishy with the connection between the FC and the RCV (they are on the [same board](http://www.overskyrc.com/scisky-micro-32b-f1-brushed-fc-receiver-optionvtx-25mw-camera-p-747.html)), but since its working just fine with BF 3.0.1, it might also be a bug.

If I can support this problem with testing something, please let me know.

sjb0095, 2017-12-02:
I have the same problem with the Warlark 80 - I'm unable to upgrade above 3.0.1 or it breaks the frsky receiver,  I've tried all the way up too 3.2.2 and no joy :(

DieHertz, 2017-12-02:
https://www.rcgroups.com/forums/showpost.php?p=38455867&postcount=130

closed 2017-12-02 by DieHertz