betaflight/betaflight #2516: Feature request: Accelerometer speed sensing

ErikEklund, 2017-02-28:
If the accelerometer is enabled in acro mode, would it not be possible to use the quad angle of the gyro and the accumulated forward acceleration to get a somewhat accurate speed reading? Having an approximate speed for the OSD without needing a GPS would be a nice addition.

RipperDrone, 2017-02-28:
It's difficult IMHO: Integrating accel to determine vel has to happen from start when v_start = 0 is a knwon starting point. But inflight, acc noise error will be accumulated and cause unpredictable deviation in the integral vs true speed. Just from my experience with automotive accel sensors and trying to derive estimates for actual speed...

Other opinions?

benjaminmcd, 2017-02-28:
In my experience with using accelerometer for speed or distance calculation is effective for a couple of seconds at best. I don't pretend to know everything but in my experience it will be inaccurate at best.

andre92, 2017-03-01:
iNav is doing just that but it requires a gps to be accurate for longer than a few seconds. There is no way to compensate for long term drift without it.

ErikEklund, 2017-03-04:
I suppose it's not possible with continuos measurement. An idea could be to "zero out" the counter through an aux channel, so that you can do a few seconds speed tests. Good for just trying speeds in a punch out or do a few seconds run to get an approximate value. Although perhaps of limited use.

RipperDrone, 2017-03-04:
When you zero out the counter you need to know the actual speed v0, otherwise integration of the accel signal only yields differential speed relative to when you zeroed. This means you have to stop the quad, be sure it's at zero horizontal speed, and then start the measurement...

ErikEklund, 2017-03-05:
You could always reset it when arming. I could still see how it could be useful for arming, trying a short full throttle run and get an approximate top speed, to have comparison numbers. But no, won't work for continuous measurement though, and would probably need to disable it after X seconds due to errors accumulating. I can definitely see how confusing that would be.

evilmerlin, 2017-04-29:
How about just getting acceleration from accelerometer?

RipperDrone, 2017-04-29:
Yes, that's ok for thrust comparison. Max speed would be nice though. Can't be calculated from accel, mathematical integration collects too much error over time...

evilmerlin, 2017-04-29:
So any idea how to go about calculating it?

RipperDrone, 2017-04-29:
no need, is transmitted over telemetry - I'd want to know speed, not accel...

ErikEklund, 2017-05-02:
Wouldn't the speed calculations from the accelerometer be somewhat accurate for at least a couple of seconds, before it would start do accumulate too much errors?

If so, I'd still opt for a "speed measure mode" triggered through an aux channel switch, where it calculates speed for the following 5-10 seconds (however long it's possible to get a somewhat accurate reading from accelerometer calculations) and then it stops, and displays the top measured speed during that time in the OSD. While it's far from as good as a constant GPS speed reading, it would be a nice feature just for testing your quad and do comparisons.

RipperDrone, 2017-05-02:
Would still require to know v0 (actual speed when you flip the mode switch), can only calculate differential speed from there

ErikEklund, 2017-05-03:
Yes, I guess the best way is to zero it on arm, and then measure for X seconds after initial throttle up. But I agree it's an awkward way to get around the limitations. As mentioned, feel free to close this if it feels to limited in usability, I'm just brainstorming here.

RipperDrone, 2017-05-03:
think you need to close it yourself...?

ErikEklund, 2017-05-16:
Closing. While technically possible, it would only work from a standstill and for a couple of seconds, making it of limited use.

closed 2017-05-16 by ErikEklund