betaflight/betaflight #2528: BF 3.1.6 on cc3d evo always in bind mode

alexandreBoussemart, 2017-03-01:
Hello,

I have flashed my cc3d evo in betaflight 3.1.6. Everything works except the satellite. I plugged it on the flexport and as soon as I turn on the card it always goes in bind mode.
Yet I put the uart3 in serial_rx
Also, is it possible to mount a buzzer hobbyking on the evo?

I need help
thank you
Alexandre

astrobond, 2017-03-02:
Same problem for me, maybe the i2c sharing the same pin on the flexiport for CC3D F1 evo ?

astrobond, 2017-03-02:
For the buzzer, never testet, but you need to add a transistor and a resistor to a standard buzzer:
https://github.com/cleanflight/cleanflight/blob/master/docs/Wiring/CC3D%20-%20buzzer%20circuit.pdf
and connect the resistor to the motor pin 6.
(and maybe type in cli "set enable_buzzer_p6 = ON")

alexandreBoussemart, 2017-03-02:
I have the buzzer hobbyking, I need the transistor and the resistance?
You are in what version of betaflight? I will try to go to 3.0.1

astrobond, 2017-03-02:
Yes 3.0.1, no problem with autobind :)
You need a 5V active buzzer (naze 32 compatible buzzer) with the transistor and the resistance.

alexandreBoussemart, 2017-03-02:
Thanks, i will try the 3.0.1
I have this one, it does not work?
https://hobbyking.com/en_us/hobbykingr-discovery-buzzer.html/?___store=en_us

AndersHoglund, 2017-03-02:
Satellite bind issue confirmed on CC3D and 3.1.6, as well as some other 3.1.x version.

There are spurious pulses on UART3 Rx line during some IO setup, seen them on the scope but not located what code is causing them. There have also been reports on RCG with other FCs and satellites stuck in bind mode, not only CC3D and CC3D-EVO. 

A simple fix is to add 200ms more delay in fc/fc_init.c and let the satellite bind window close before continuing with IO setup. There are already a 100ms delay, but that is not enough and might also be there for other reasons not related to spektrum binding. I can prepare a PR for a quick fix like that. 

See attached picture of the first pulse. There are more later on, but too thin to be visible on a screen shot.... 
[image]

astrobond, 2017-03-02:
@alexandreBoussemart you can try your "discovery-buzzer" without transistor and resistor, but i'm not sure it will work...
@AndersHoglund thank's a lot for the fix :)

alexandreBoussemart, 2017-03-02:
@AndersHoglund Thank's, To add your patch it is enough that I reflash my map by reloading version 3.1.6?

AndersHoglund, 2017-03-03:
Builds with PR 2534 integrated can be downloaded here: http://andwho.sytes.net:8080/job/BorisB_BetaFlight_Maint/

alexandreBoussemart, 2017-03-10:
Hello,
I tried your version and it works perfectly.
thank you very much

closed 2017-03-15 by AndersHoglund

alexandreBoussemart, 2017-03-15:
Hello, 
Your version contains the enable_buzzer_p6? When I do CLI Command: set enable_buzzer_p6 = ON 
I have an error invalid name.
thank's

AndersHoglund, 2017-03-16:
No. That is an old thing now gone. Use resource re-mapping instead. 

    # resource motor 6 none
    Freed
    # resource beeper 1 a02
    A02 set
    # save

In what way is this related to the Spektrum bind issue?

alexandreBoussemart, 2017-04-06:
Hello,

I now have several flights with your resource and an orange satellite dsm2 R100 and everything works fine.
I wanted to mount a 130 with this satellite: https://hobbyking.com/fr_fr/orangerx-r111xn-dsmx-dsm2-compatible-nano-satellite-receiver.html
However as soon as I plug the card to betaflight with the satellite can not detect the card. If I plug the card without satellite then connect it to betaflight, as soon as I reconnect the satellite it automatically disconnects the card. Do you have any idea?

Thank you

AndersHoglund, 2017-04-06:
If the FC works with the R100 sat, but not with the R111 then it is most likely NOT any BetaFlight issue.
I would guess you draw too much current and there is a FC brown out, maybe due to a wrong connection. Hard to tell without the gear at hand. Check your LEDs and voltages, 5V and 3.3V.

alexandreBoussemart, 2017-04-06:
I looked and I have 3.3v at the exit.
As soon as I plug it in place of the other satellite, there is only the green LED of the board that lights up and the satellite led remains fixed red (with radio on)