betaflight/betaflight #2531: Gyro problem on Omnibus F3 1.1?

simonsais, 2017-03-02:
Hey there! I recently had an issue on my new Omnibus F3 1.1 setup which flew fine, and from one lipo to another it had serious roll oscillations. I changed props, then motors, then escs with Multi and Dshot and still got those oscillations. 

https://www.dropbox.com/s/l0puog6lifqy3td/flying%20oscillations.BFL?dl=0

After that i disassembled the craft, and tried to make a blackbox log not in flight, only by rolling and pitching by hand. 

https://www.dropbox.com/s/so7itqtkhc3i6jh/hand%20moved.BFL?dl=0

Could it be a min_throttle thing? I hope somebody understands what i didn't...

I ran BF 3.1.6 with following settings:

# Betaflight / OMNIBUS 3.1.6 Feb 21 2017 / 14:33:10 (1fd502c)
name -
resource BEEPER 1 C15
resource MOTOR 1 B08
resource MOTOR 2 B09
resource MOTOR 3 A03
resource MOTOR 4 A02
resource MOTOR 5 B07
resource MOTOR 6 B06
resource MOTOR 7 NONE
resource MOTOR 8 NONE
resource MOTOR 9 NONE
resource MOTOR 10 NONE
resource MOTOR 11 NONE
resource MOTOR 12 NONE
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 B04
resource PWM 1 NONE
resource PWM 2 NONE
resource PWM 3 NONE
resource PWM 4 NONE
resource PWM 5 NONE
resource PWM 6 NONE
resource PWM 7 NONE
resource PWM 8 NONE
resource LED_STRIP 1 A08
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A14
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 NONE
resource SERIAL_TX 5 NONE
resource SERIAL_TX 6 NONE
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_TX 11 NONE
resource SERIAL_TX 12 NONE
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 A15
resource SERIAL_RX 3 B11
resource SERIAL_RX 4 NONE
resource SERIAL_RX 5 NONE
resource SERIAL_RX 6 NONE
resource SERIAL_RX 7 NONE
resource SERIAL_RX 8 NONE
resource SERIAL_RX 9 NONE
resource SERIAL_RX 10 NONE
resource SERIAL_RX 11 NONE
resource SERIAL_RX 12 NONE
mixer QUADX
[... 416 lines cut ...]
set p_alt = 50
set i_alt = 0
set d_alt = 0
set p_level = 50
set i_level = 50
set d_level = 100
set p_vel = 55
set i_vel = 55
set d_vel = 75
set level_sensitivity = 55
set level_limit = 55
rateprofile 0
rateprofile 0

set rc_rate = 166
set rc_rate_yaw = 186
set rc_expo = 6
set rc_yaw_expo = 6
set thr_mid = 50
set thr_expo = 0
set roll_srate = 76
set pitch_srate = 76
set yaw_srate = 76
set tpa_rate = 10
set tpa_breakpoint = 1650

quadpaul, 2017-03-10:
I checked your black box logs I could not see anything that stuck out to me, how are you powering your bored? I think it might be a bad 5v power supply...

simonsais, 2017-03-10:
thanks for having a look! i power up through the vbat pins - the f3 omnibus uses a onboard regulator. maybe i'll try the 5v output of my matek pdb and see if there was a problem! :)

closed 2017-10-23 by jflyper