betaflight/betaflight #2532: Softserial issue with TBS Unify Pro HV Race

mhenrichsen, 2017-03-02:
Info:
Holybro Kakute v1 (not AIO) running SP racing 3.1.6 software
Audio is soldered to Motor 6 pad.

Standard resource mapping:

```
resource BEEPER 1 C15
resource MOTOR 1 A06
resource MOTOR 2 A07
resource MOTOR 3 A11
resource MOTOR 4 A12
resource MOTOR 5 B08
resource MOTOR 6 B09
resource MOTOR 7 A02
resource MOTOR 8 A03
resource PPM 1 A00
resource PWM 1 A00
resource PWM 2 A01
resource PWM 3 B11
resource PWM 4 B10
resource PWM 5 B04
resource PWM 6 B05
resource PWM 7 B00
resource PWM 8 B01
resource LED_STRIP 1 A08
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A14
resource SERIAL_TX 3 B10
resource SERIAL_TX 11 B05
resource SERIAL_TX 12 B01
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 A15
resource SERIAL_RX 3 B11
resource SERIAL_RX 11 B04
resource SERIAL_RX 12 B00
```
My resource mapping: (changed serial tx 11 to B09)

```
resource BEEPER 1 C15
resource MOTOR 1 A06
resource MOTOR 2 A07
resource MOTOR 3 A11
resource MOTOR 4 A12
resource MOTOR 5 B08
resource MOTOR 7 A02
resource MOTOR 8 A03
resource PPM 1 A00
resource PWM 1 A00
resource PWM 2 A01
resource PWM 3 B11
resource PWM 4 B10
resource PWM 5 B04
resource PWM 6 B05
resource PWM 7 B00
resource PWM 8 B01
resource LED_STRIP 1 A08
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A14
resource SERIAL_TX 3 B10
resource SERIAL_TX 11 B09
resource SERIAL_TX 12 B01
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 A15
resource SERIAL_RX 3 B11
resource SERIAL_RX 11 B04
resource SERIAL_RX 12 B00
```
Settings dump
[... 472 lines cut ...]
set d_vel = 75
set level_sensitivity = 55
set level_limit = 55
rateprofile 0
rateprofile 0

set rc_rate = 100
set rc_rate_yaw = 100
set rc_expo = 0
set rc_yaw_expo = 0
set thr_mid = 50
set thr_expo = 0
set roll_srate = 70
set pitch_srate = 70
set yaw_srate = 70
set tpa_rate = 10
set tpa_breakpoint = 1650
```
Telemetry work just fine, just not getting any input from TBS Unify Pro.

Under Dev on the Taranis page it usually says "255 or None", just says "---" with this setup.

Any ideas?

Thanks!

jflyper, 2017-03-02:
B9 shares timer TIM4 with MOTOR 3 (A11) and MOTOR 4 (A12).

Use TX side of the receiver UART (I presume it's on UART3, and labelled RX3), which is on B10.

```
resource SERIAL_TX 3 none
resource serial_tx 11 b10
```

Also, you should NOT enable multiple ports enabled for SmartAudio; turn it off for SOFTSERIAL2.

mhenrichsen, 2017-03-02:
Hey jflyper!

Thanks for the great work! 

Seems like I got a bit further, now I'm getting this on my Taranis, but only on the vTX page. PIDS and so on still work fine.
[image]

Any ideas?

mhenrichsen, 2017-03-02:
I updated to the newest files:
Same error just at line 386 instead.

```
local function updateVTXFreq(page)
   page.values["f"] = freqLookup[page.values[2]][page.values[3]]
end
```

mhenrichsen, 2017-03-02:
I found the issue.
If your TBS Unify Pro HV is on the L band, you will get the above issue.
The fix: Change band to something other than the L band.

@jflyper Can you update the file, so it can take the L band?
The frequensies are:
5362, 5399, 5436, 5473, 5510, 5547, 5584, 5621.

Thanks a lot for your help. Do you have anywhere I can donate or something?

jflyper, 2017-03-03:
@mhenrichsen Thanks for the investigation.

I'll look for a way to enable Low Race band on Unify; there are couples of things we have to consider.

Positive:
- It's easy to add a LORACE band/channel data.

Negative (technical):
- HV and HV Race are not distinguishable from software; if we just add LORACE and let users to pick a channel in LORACE, then there's nothing preventing non-Race users from attempting, and they will be complaining. Should be some way to stop this from happening.

Negative (compliance):
- I'm wondering if frequencies in LORACE can be used legally by amateurs in part of the World where majority of the BF users resides. For existing bands. which already have established as _de facto_, I have explicitly wrote a note in the BF wiki that compliance to local regulation is user's responsibility, but for the LORACE, I'm wondering if I should even enable it.

Negative (political):
- I don't know how broadly the LORACE is accepted by the community. With current implementation, adding  band/channel data will enable LORACE for other vtx also, which may benefit the particular commercial entity.

jflyper, 2017-03-03:
@mhenrichsen You can always press the Donate button at
[https://www.rcgroups.com/forums/showthread.php?2464844-Betaflight-Flight-Controller-Firmware-Discussion-Thread](https://www.rcgroups.com/forums/showthread.php?2464844-Betaflight-Flight-Controller-Firmware-Discussion-Thread)

mhenrichsen, 2017-03-03:
@jflyper 

> Positive:
> 
> It's easy to add a LORACE band/channel data.
Great. I've actually tried to add them myself, and I can switch channels on my Taranis, but they when I save it switches back to the channel/band I was on before.

> Negative (technical):
> 
> HV and HV Race are not distinguishable from software; if we just add LORACE and let users to pick a channel in LORACE, then there's nothing preventing non-Race users from attempting, and they will be complaining. Should be some way to stop this from happening.

That is true, but it could be possible to add a text when they select it? Something like "Only for Unify Pro Race"?

> Negative (compliance):
>
> I'm wondering if frequencies in LORACE can be used legally by amateurs in part of the World where majority of the BF users resides. For existing bands. which already have [...]

Transmitting at over 200mw is illigal most places too, and you have to actively activate it on the Unify. When you active it you get a bunch of other channels too. It must be up to the user to use the device as intented?

IdefixRC, 2017-03-06:
Hi guys,
 
I'm trying to get this to work with my SPRacing F3 Board.
First things first: When I connect SmartPort to UART1 and SmartAudio to UART2 it all works fine (when armed). Now I enabled SmartPort on UART2 and I'm trying to get SmartAudio to work with SoftSerial 1.
PIDs work fine, no data from the Unify :-(

Reading the above I set PPM and PWM1 to none (A0 before) and configured serial TX 11 to A0.
Unfortunately no luck.

To rule out that I messed the PIN's up: After moving Serial TX 11 to A0 I moved the audio line to PPM (IO_1 Pin3 which is labled PIN 3 CH1 / PPM in the manual)

Picture of the Board and Pins: [http://www.rjxhobby.com/image/data/SP%20RACING%20F3-1.jpg](http://www.rjxhobby.com/image/data/SP%20RACING%20F3-1.jpg)

Here is the CLI Dump & Thanks for the Help !

`# Betaflight / SPRACINGF3 3.1.6 Feb 21 2017 / 14:44:08 (1fd502c)
name -
resource BEEPER 1 C15
resource MOTOR 1 A06
resource MOTOR 2 A07
resource MOTOR 3 A11
resource MOTOR 4 A12
resource MOTOR 5 B08
resource MOTOR 6 B09
resource MOTOR 7 A02
resource MOTOR 8 A03
resource MOTOR 9 NONE
resource MOTOR 10 NONE
resource MOTOR 11 NONE
resource MOTOR 12 NONE
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 NONE
resource PWM 1 NONE
resource PWM 2 A01
resource PWM 3 B11
resource PWM 4 B10
resource PWM 5 B04
resource PWM 6 B05
resource PWM 7 B00
resource PWM 8 B01
resource LED_STRIP 1 A08
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A14
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 NONE
resource SERIAL_TX 5 NONE
resource SERIAL_TX 6 NONE
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_TX 11 A00
resource SERIAL_TX 12 B01
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 A15
resource SERIAL_RX 3 B11
resource SERIAL_RX 4 NONE
resource SERIAL_RX 5 NONE
resource SERIAL_RX 6 NONE
resource SERIAL_RX 7 NONE
resource SERIAL_RX 8 NONE
resource SERIAL_RX 9 NONE
resource SERIAL_RX 10 NONE
[... 406 lines cut ...]
set d_alt = 0
set p_level = 50
set i_level = 50
set d_level = 100
set p_vel = 55
set i_vel = 55
set d_vel = 75
set level_sensitivity = 55
set level_limit = 55
rateprofile 0
rateprofile 0

set rc_rate = 125
set rc_rate_yaw = 125
set rc_expo = 30
set rc_yaw_expo = 30
set thr_mid = 50
set thr_expo = 0
set roll_srate = 70
set pitch_srate = 70
set yaw_srate = 70
set tpa_rate = 10
set tpa_breakpoint = 1650

# `

jflyper, 2017-03-06:
@IdefixRC SS half-duplex works with PPM for SPRF3.

https://github.com/betaflight/betaflight/wiki/Single-Wire-Software-Serial

Do you have something connected and enabled on RC_CH2?

IdefixRC, 2017-03-06:
Thanks for the quick reply. 
I have Smartport on UART2 and serial rx on UART3. Aside from the 4 motors and a buzzer nothing else is connected to the FC. 

Thanks for the help

jflyper, 2017-03-06:
Can you backup your configs and test with fresh flash (or execute cli defaults), connecting only the smartaudio on ppm? You may want to try other pins listed in the above wiki page.

IdefixRC, 2017-03-07:
Ok.will try and report back. Thanks

IdefixRC, 2017-03-07:
Tested PPM, PWM1 and PWM2. No luck.
Moved it to motor 8 (A03) and it worked immediately. looks like the IO ports might not work for the SPRacingF3 board. Unless someone else can confirm their function I would probably mark them with a ? in the wiki. Thanks for the help @jflyper  !!

wolverin3, 2017-06-11:
# diff
# Betaflight / BETAFLIGHTF3 3.1.7 Apr  3 2017 / 21:43:25 (e1c4b5c)
name Wolverine
resource PPM 1 NONE
resource SERIAL_TX 11 B07

feature -BLACKBOX
feature -SDCARD
feature SOFTSERIAL
feature LED_STRIP
feature AIRMODE
serial 0 32 115200 57600 0 115200
serial 30 2048 115200 57600 0 115200
led 0 4,2::COW:3
led 1 5,2::COW:3
led 2 6,2::COW:3
led 3 7,2::COW:3
led 4 8,2::COW:3
led 5 9,2::COW:3
led 6 10,2::COW:3
led 7 11,2::COW:3
aux 0 0 0 1300 2100
aux 1 4 0 1300 2100
aux 2 13 2 1700 2100
set rssi_channel = 8
set fpv_mix_degrees = 30
set motor_pwm_protocol = DSHOT300
set acc_hardware = NONE
set pid_process_denom = 1
set osd_units = METRIC
set osd_rssi_alarm = 45
set osd_cap_alarm = 1200
set osd_time_alarm = 3
set osd_vbat_pos = 2472
set osd_rssi_pos = 2490
set osd_flytimer_pos = 2518
set osd_ontimer_pos = 374
set osd_throttle_pos = 2497
set osd_horizon_pos = 200
set osd_current_pos = 2465
set osd_mah_drawn_pos = 2480
set osd_craft_name_pos = 2504
profile 0

set d_lowpass_type = PT1
set p_pitch = 63
set i_pitch = 45
set d_pitch = 25
set p_roll = 49
set i_roll = 35
set d_roll = 20
rateprofile 0
rateprofile 0

set rc_rate = 150
set rc_rate_yaw = 130
set rc_expo = 15

sadownik, 2017-07-22:
@jflyper 

how difficult would it to add Lorace frequencies to betaflight and the lua script? Could you provide updated files for people compiling the target on their own?