betaflight/betaflight #2535: Compilation error in  "pwm_output_stm32f7xx.c",

Plantgrower, 2017-03-02:
I am using the MBED platform to port Dshot for my own use in a project. I am down to one last error. Could someone suggest why I am seeing this error and how to solve it.

Error: Argument of type "std::uint8_t" is incompatible with parameter of type "dmaIdentifier_e" in "pwm_output_stm32f7xx.c", Line: 164, Col: 14

This is the line of code
    dmaInit(timerHardware->dmaIrqHandler, OWNER_MOTOR, RESOURCE_INDEX(motorIndex));

borisbstyle, 2017-03-03:
can you make it uint8_t instead of std::uint8_t ?

Plantgrower, 2017-03-03:
Do you mean make timerHardware->dmaIrqHandler uint8_t ? 
This is already declared as uint8_t . I have tried casting it to uint8_t with no luck.

martinbudden, 2017-03-03:
Did you try casting it to `dmaIdentifier_e`?

Plantgrower, 2017-03-04:
@martinbudden  Thanks that worked. I have given up on using mbed for now and am using eclipse and command line compiling.

closed 2017-10-23 by jflyper