betaflight/betaflight #2537: F4 OSD Menu Bug

MiddleMan5, 2017-03-03:
Issue Summary: Navigating to the SCR layout then backing up to main menu, then scrolling to the next page causes OSD to jump back to scr layout and is un-exitable. Craft power cycle required.

Details: Happens on target OMNIBUSF4SD with OSD enabled. Detected in both 3.1.5 and 3.1.6.

Reproducible? Yes, I've only noticed it when setting SCR layout
Reproduction:
Power on Craft, connect headset.
Enter BF menu (yaw left, throttle mid, pitch up)
Enter SCR Layout section of menu.
Scroll down to next page.
Go back to main menu.
Scroll down past bottom of page, menu will jump back to page 2 of SCR Layout.
You should now be unable to exit the osd functionality, reboot is required to clear locked state.

I'm assuming this has something to do with how betaflight caches the osd pages. My assumption is that if a menu category doesn't have another page, but you scroll anyway, it just pulls the last cached page.

Sorry for the formatting, I'm typing this on my phone.

jflyper, 2017-03-03:
Yes. This is one of the forgotten item in the TODO list for CMS.

The original OSD code which the CMS was derived from was only capable of handling two pages at the most (fixed coding based on an assumption that a menu never exceeds two pages long).  I realized this during the CMS development, but took make it work and improve later approach, and never improved it.

The limitation/bug is acknowledged now. I'm going to fix it.

@martinbudden Can you add a BUG label?

martinbudden, 2017-03-03:
Done.

martinbudden, 2017-03-03:
@jflyper , don't you have write access to the github repository?

jflyper, 2017-03-03:
@martinbudden I don't think so ...

borisbstyle, 2017-03-14:
@jflyper should this go into 3.1.7 as a fix?

Fauropitotto, 2017-04-04:
Was this issue ever revisited?

jflyper, 2017-04-05:
It's in my todo list, but priority has been kept low; thought it's not fatal.
But as it has been a month, I'll bumped it up and give it a look.

jflyper, 2017-04-06:
@MiddleMan5 
I have a PR candidate. Can you test your case (and other cases if possible)?

https://github.com/jflyper/cleanflight/tree/bfdev-cms-refactor

jflyper, 2017-04-27:
@MiddleMan5 
Do you need a firmware with this fix to test?

MiddleMan5, 2017-04-27:
Sorry I haven't gotten back to you! My F4 quad is down at the moment, but yeah if that previous link still has the patch candidate then I'll see if I can throw it on the board, today or tomorrow  most likely

MiddleMan5, 2017-04-29:
Okay, sorry about the delay,

yes, I have confirmed that does fix that specific issue, however now there's another one:
I haven't tested other cases, but when I go to the pid menu and either go one menu item above the top OR all the way past the bottom, the osd crashes and disappears.

MiddleMan5, 2017-04-29:
I have tested a few other cases, preliminary observations show just the pid menu crashing. I also have a few other strange behaviors such as the voltage icon showing twice and the throttle stick disabling input unless exactly in the center (annoying)

jflyper, 2017-05-01:
@MiddleMan5 Thanks for testing. I'll look into the issues you've reported.

jflyper, 2017-05-01:
@MiddleMan5 

- The PID page issue was fixed (Incorrect page count calculation); PR has been updated.
- Second battery icon is for average cell voltage.
Configurator may not have a enable/disable switch for this (CMS does).
- The throttle position was too loose in the previous settings of CMS.
It was made more strict to avoid stick combinations interpreted as other commands.

martinbudden, 2017-08-01:
Fixed byhttps://github.com/betaflight/betaflight/pull/2833

closed 2017-08-01 by martinbudden