betaflight/betaflight #2542: BF 3.1.6 TBS Colibri Race 1.2 isn't recognized anymore

madjan06, 2017-03-03:
I have three quads with a tbs colibri race 1.2 on them, when I try flashing bf 3.1.6 the FC isn't recognized anymore by the Betaflight Configurator. There is no com-port showing up and my windows 10 shows the message thats something wrong with the connected usb-device. When I flash back 3.1.5 everything works fine. I have tested it on all of my three FCs, so it seems to be a colibri related issue.

mikeller, 2017-03-05:
I cannot verify / test this because I don't have the hardware to test. But this could be a case of a faulty configuration being used after the upgrade. Can you try flasing with 'Full chip erase' being checked, and see if this makes 3.1.6 work for you?

madjan06, 2017-03-05:
Hi,
I tried the upgrade with 'Full chip erase' enabled. But I still get the same behavior, then I flash back 3.1.5 and everything works fine. I think there has something changed in 3.1.6 that causes this Problem.

AndersHoglund, 2017-03-05:
The only change from 3.1.5 to 3.1.6 that are specific to COLIBRI_RACE are two soft serial ports added. 
Then of course there are all the common updates...

    $ git checkout v3.1.6
    HEAD is now at 1fd502c... Lower F1 looptime restriction
    $ git diff v3.1.5 src/main/target/COLIBRI_RACE/target.h
    diff --git a/src/main/target/COLIBRI_RACE/target.h b/src/main/target/COLIBRI_RACE/target.h
    index 7fe5962..42d28fd 100755
    --- a/src/main/target/COLIBRI_RACE/target.h
    +++ b/src/main/target/COLIBRI_RACE/target.h 
    @@ -80,7 +80,10 @@
     #define USE_UART1  
     #define USE_UART2
     #define USE_UART3
    -#define SERIAL_PORT_COUNT       4
    +#define USE_SOFTSERIAL1
    +#define USE_SOFTSERIAL2
    +
    +#define SERIAL_PORT_COUNT       6

Try reverting this and rebuild. I have no FC to test...

madjan06, 2017-03-05:
After I got docker running, I tried your suggestion. I commented
`+#define USE_SOFTSERIAL1
+#define USE_SOFTSERIAL2` 
and changed SERIAL_PORT_COUNT  back to 4. 

It seems to fix the problem, I flashed my build and the FC is recognized as usual. So the Colibri seems to have an issue with the SOFTSERIAL. I hope this has no other effect on other parts of the firmware. Maybe you can fix this for the next Release.

Thanks for your support, you made my day ;-)

jflyper, 2017-03-05:
@madjan06 What if BST is dropped? That is, without `USE_BST` but with `USE_SOFTSERIALx`.
```
//#define USE_BST
```

madjan06, 2017-03-05:
@jflyper Just for verification, I dropped BST and tried it again. It also works, I had no time to check if all BST-Devices like the core pro or the fpvision are still working.

jflyper, 2017-03-06:
@madjan06 Yes. If you drop the USE_BST, the bst will not be working (naturally) ;)
I just wanted to narrow down where the real problem is.
Your test result confirmed that it is not the software serial alone that is preventing the board from being recognized, but there is something that BST support code that is assuming there is no software serials.

jflyper, 2017-03-06:
I think the immediate solution would be to drop (disable) the software serial for the moment.

Right now, I will submit a PR for COLIBRI_RACE target to drop the software serials.

If TBS folks feel the software serial is required for Colibri Race, they can trouble shoot the BST support code and override the PR.

madjan06, 2017-03-06:
@jflyper I agree with you, thanks for creating the pull request, so everything should be fine in the next release. 
Thanks for the quick response, I really appreciate the work you guys put into that project ;-)

closed 2017-04-06 by madjan06