betaflight/betaflight #2548: Betaflight was connecting.  Now it's not?

Steve1Graham, 2017-03-04:
Hey guys.  I'm struggling to get betaflight to reliably connect to my macbook.  A few weeks ago I bought a Dodo FC and was unable to connect it to my macbook.  I tried a number of the solutions suggest here but to no avail.  I took it to a friends house who has more experience than I and he was having trouble getting it to connect as well.

A few days ago I acquired a beebrain for my tiny whoop and it connected fine.  I did the firmware flash and setup and have been flying that unit.  At some point I got the idea to try the dodo again and lo and behold it worked fine.  I flashed the firmware and spent some time poking around in the software.  I must have connected and disconnected the beebrain and dodo 20 times or more and it was working fine.  During this process I recorded the port the units were connecting on:  /dev/cu.SLAB_USBtoUART

Tonight I tried to connect the dodo and it opens a serial port then closes it again about 10 seconds later.  The port address I listed above is not auto populating so I tried to manually connect by typing it in but that did not work either.

What could cause a perfectly functioning connection one day to suddenly stop working the next or vice versa?

martinbudden, 2017-03-04:
Are you using the same USB cable? Some USB cables that come with chargers only contain the power wires.

Steve1Graham, 2017-03-04:
I think it's an issue with the USBc port on this macbook.  I have a little dongle that plugs into the computer that gives me expansion ports for regular USB and and SD card.  The connection with this poorly designed peripheral is intermittent so I have to brace it so the connection is reliable.  I got an extra USBc to USBc that would allow me to plug just a cable into the computer and have the adaptor at the end to see if the connection would be more reliable.  It powers up the FC just fine but clearly has a problem communicating in this configuration because as soon as I went back to plugging the adaptor directly into the computer it found the port that I listed above and connected just fine.  Very strange and far above my paygrade to even begin to figure out why it would only work this way.

I think we may just be outsmarting ourselves with new connection protocols every 6 months.  More often than not they create more problems than solutions.  :(

Thanks.

martinbudden, 2017-03-04:
Yes, a direct connection to the macbook is the most reliable. Closing this issue since it is hardware related.

closed 2017-03-04 by martinbudden