betaflight/betaflight #2551: Nucleo F7 which pins carry the PWM signal.

Plantgrower, 2017-03-04:
I am using a Nucleo F767ZI board. Which pins carry the motors PWM signal, and how is that mapped out ?

jflyper, 2017-03-04:
You can flash the board with NUCLEOF7 target and ask CLI with `resource list` and `resource`.

For timer association, refer to
https://github.com/betaflight/betaflight/blob/v3.1.6/src/main/target/NUCLEOF7/target.c

Note that timer/pin allocation changes from time to time. Always check when you flash something new.

Plantgrower, 2017-03-04:
@jflyper Thanks. The board has a built in ST-link  v2.1 . I can't get the configuration app to see the board.
Do you know what is the best way to flash and communicate with the board using the app and cli ?

jflyper, 2017-03-04:
I connected the OTG USB (the one on the target side) as well as STLINK side USB for power, to a single hub, and shorted BOOT0 to 3V3 while powering up the whole board by powering the hub.

This throws F7 into DFU.

Plantgrower, 2017-03-04:
@jflyper Thanks for the info and the incredibly fast response ! Am I right in thinking I will have to set #define STM32F767xx  to compile for my particular board ?

jflyper, 2017-03-04:
Oh, yeah, I believe you should. Overlooked it caz I'm on F746ZG part ;)

Plantgrower, 2017-03-05:
@jflyper  I have managed to flash the board with cleanflight. When I try to connect with cleanflight all I get is " Failed to open serial port". What am I doing wrong ? The data usb is connected target side and powered by the stlink side. Do I need udev rules adding ?

Plantgrower, 2017-03-05:
@jflyper I am using Ubuntu btw.

Plantgrower, 2017-03-05:
@jflyper Added udev and can now connect to the board. The date the new version was released is displayed but it then gives the error "No configuration received within 10 seconds, communication failed". The only change I made was to add #define STM32F767xx to the top of main.c and then compiled with make TARGET=NUCLEOF7 .

mikeller, 2017-03-06:
What is the device that your board is recognised as? If it is `/dev/ttyACM<X>`, then you need to disable modemmanager, or it will claim the device.

closed 2017-10-23 by jflyper