betaflight/betaflight #2554: Blackbox (no option to save log files) in configurator

sa-mendez, 2017-03-04:
I am using the following with the 1.9.3 version of the Betaflight configurator:

# version
# Betaflight / OMNIBUSF4SD 3.1.6 Feb 21 2017 / 14:34:26 (1fd502c)

I have it setup to use Blackbox and to log to the On-borard SD card slot, yet when I go to the Blackbox sectionI do not have any option to download the logs, or even clear out the SD card.

I know that logging is working, because if I pull out the SD card and put it into an SD card reader, I see the logs are being created.  I have downloaded them and been able to use the viewer to look through them.

Is this a problem with something I forgot to setup, or is this a known issue?

Thanks in advance,

Sergio

Here is a screenshot of what I see:

[image]

unitware, 2017-03-04:
Yep, there is no download from SD-card. You need to remove the card and put it in a computer. I'd love to be able to do this though - my SD-card is hard to reach.

sa-mendez, 2017-03-04:
Yeah, I see the source of my confusion now.  The options to erase/download are only available when using a dataflash chip.  I mixed the two settings up.

Now, if it were possible to implement this, I would be very grateful ;-)

mikeller, 2017-03-05:
Considering how slow dataflash download is, and how space there is on an SD card, I don't know how much use such a facility would be. But definitely an option for F4, where there is plenty of flash space.

robert-hoffmann, 2017-08-22:
+1
I don't care if sdcard download through a passthrough is slow, i mean for some quick tests, a couple of logs could be just a couple hundred KB, but considering how painful it is to insert/remove most sdcards is ..the extra latency is probably worth it !

jasonhyland, 2018-01-27:
+1 for pain in the backside sd cards

momo2xlc, 2018-02-08:
+1 cant access SD card through, would be great to get download option