betaflight/betaflight #2558: NucleoF7 "No configuration received within 10 seconds, communication failed"

Plantgrower, 2017-03-05:
This is on a Nucleo F767ZI board. I compiled betaflight with the only change to the code was to add  #define STM32F767xx to the top of main.c and then compiled with make TARGET=NUCLEOF7 . I successfully flashed this to the board using cleanflight. When I try to connect to in cleanflight in the cf tab, it connects and displays the time the hex was compiled but it then gives the error "No configuration received within 10 seconds, communication failed".

Plantgrower, 2017-03-05:
The log is as follows:-

2017-03-05 @ 11:43:41 -- Serial port successfully opened with ID: 7
2017-03-05 @ 11:43:41 -- MultiWii API version received - 1.32.0
2017-03-05 @ 11:43:41 -- Flight controller info, identifier: BTFL, version: 3.2.0
2017-03-05 @ 11:43:41 -- Running firmware released on: Mar 5 2017 06:41:39
2017-03-05 @ 11:43:51 -- No configuration received within 10 seconds, communication failed

jflyper, 2017-03-06:
I found out there is no code active for `STM32F767xx`, so there may be a code you have to add/delete for `STM32F767xx`.

Since you have the STLINK, why don't you tap into it and see what it's doing after giving the firmware information?

Plantgrower, 2017-03-07:
@jflyper Do you have any links to articles on how to do this and what information I am looking for.

jflyper, 2017-03-07:
@Plantgrower I don't have any pointers to refs atm as far as part of the code you have to add/delete. My suggestion was based on my experience from last few month, fiddling with timers, interrupts dmas and others.

For ST-LINK, I think there is a wiki for how to do this with VisualGDB...
Ah, this is it. By @blckmn 👍 
[Setting up VisualGDB for building Betaflight](https://blck.mn/2016/07/setting-up-visualgdb-for-building-betaflight/)

closed 2017-10-23 by jflyper