betaflight/betaflight #2561: facing issues with reading two MSP serial ports at the same time.

Abu-Abdullah, 2017-03-05:
Hi,

I'm trying to read MSP protocol from UART1 (USB) and UART2.

USB port is OK and i can access the configurator. but reading the UART2 causes unreadable MSP protocol.

a lot of bytes that does not start with $M>. some time receiving 4 bytes only "$M>1"

many output $M> without any request $M<

DODO FC, UART1-USB to Windows PC (configurator), UART2 to Raspberry Pi UART pins

another issue is the output in the CLI in the configurator, some unknow bytes is reflecting there. 

it seems one port is affecting the other one.

[image]

Abu-Abdullah, 2017-03-05:
ok, after some troubleshooting, it seems that the bytes are received in chunks and not as full response as in the USB-UART.

I'm not sure about the configurator behavior

jflyper, 2017-03-06:
You can enable MSP on multiple serial ports (max two in the past, now three on some (all?) targets, iirc).

The problem you are looking at is the behavior of MSP when one of the ports goes into CLI. When one of the MSP ports goes into CLI, rest of the ports will go into CLI also.

[https://cloud.githubusercontent.com/assets/2059463/23588811/614e30fe-01dc-11e7-8cf9-e9f10a6ae652.PNG](https://cloud.githubusercontent.com/assets/2059463/23588811/614e30fe-01dc-11e7-8cf9-e9f10a6ae652.PNG)
This is observed because the configurator connected through USBVCP is trying to talk to the board using MSP.