betaflight/betaflight #2565: LUX V2 Smartport Telemetry not working 3.1

adamreturns, 2017-03-05:
I am setting up the LUX V2 with dedicated uart 5 for telemetry and even with the switch turned on and smart port enabled on uart 5 it is not outputting any VFAS information.

set beeper_inversion = ON
set beeper_od = OFF
set serialrx_provider = SBUS
set sbus_inversion = ON
set spektrum_sat_bind = 0
set spektrum_sat_bind_autorst = 1
set tlm_switch = ON
set tlm_inversion = ON
set sport_halfduplex = ON
set frsky_default_lat =  0.000
set frsky_default_long =  0.000
set frsky_gps_format = 0
set frsky_unit = IMPERIAL
set frsky_vfas_precision = 0
set frsky_vfas_cell_voltage = OFF
set hott_alarm_int = 5
set pid_in_tlm = OFF
set bat_capacity = 0
set vbat_scale = 110
set vbat_max_cell_voltage = 43
set vbat_min_cell_voltage = 33
set vbat_warning_cell_voltage = 35
set vbat_hysteresis = 1
set ibat_scale = 400
set ibat_offset = 0
set mwii_ibat_output = OFF
set current_meter_type = ADC
set battery_meter_type = ADC
set bat_detect_thresh = 55
set use_vbat_alerts = ON
set use_cbat_alerts = OFF
set cbat_alert_percent = 10

mikeller, 2017-03-06:
> it is not outputting any VFAS information

Is it giving you any telemetry data at all?

grahamwho, 2017-03-07:
I've got a Luxv2 and s.port telem is working fine for me on uart 5.

adamreturns, 2017-03-08:
no but ive changed to a new Rx and is working perfectly so must have been that.

adamreturns, 2017-03-08:
the issue turned out to be the Rx I swapped it outa and telemetry is now working :)

martinbudden, 2017-03-08:
Closing as is a hardware issue.

closed 2017-03-08 by martinbudden