betaflight/betaflight #2566: USB port broken - need to create a custom version with MSP enabled on UART1

FPVlog, 2017-03-05:
Hello,

I broke the usb port on my omnibus F3 FC so I flashed it using a FTDI adapter via UART1 to the only version of the firmware that has MSP enabled on UART1 which is BF 3.0.1.

Now, I'd like to use the new versions of BF (3.1.6) but for the Omnibus target, MSP is not enabled on UART1.

Can someone explain to me how to change this setting in the config file and how to create a new hex file that I can flash strait away on my board?

Thank you in advance.

Cheers

mikeller, 2017-03-06:
You need to build your own custom version of the firmware, with MSP on UART1 enabled by default. See #2144 for instructions.

FPVlog, 2017-03-06:
Ok. Thanks for that. 

"So, with a broken USB; Build a custom binary with "USE_MSP_UART" and flash it over UART1. Done."

I understand what this means but I have no idea how to do it. 
Is there instructions somewhere that explain how to build a custom binary and where to put this "USE_MSP_UART" line in the code?

I'm happy to learn by myself. It's just hard to find info to start. 

Cheers

Thanks

mikeller, 2017-03-06:
Probably the easiest way to build is by using the build environment in a docker container: https://github.com/betaflight/docker-betaflight-build

With this, in your case, doing

```
docker run -e "PLATFORM=ONMIBUS" -e "OPTIONS=USE_MSP_UART" --rm -ti -vpwd:/opt/betaflight betaflight/betaflight-build
```

should do the trick.

FPVlog, 2017-03-06:
wow! Ok!

I didn't think it would be that easy!
I'll do it and let you know if I succeded.

Cheers

FPVlog, 2017-03-06:
Ok. So it worked. i've been able to build BF 3.2 for OMNIBUS but now I'd like to build version 3.1.6.

How can I do that?

Cheers

FPVlog, 2017-03-06:
All good! I found how to do it.

I downloaded the binaries and replaced all the files in my folder and I have been able to create the hex file.

I flashed the board and it works perfectly on 3.1.6!!
First time I do such a thing and I am pretty happy I did it myself!

Thanks for your help!! Thank you very much!

mbernson, 2017-03-06:
I had the same issue today, also with an omnibus F3 😞. Building a custom version with `USE_MSP_UART` worked like a charm though! I'll see if I can add these instructions to the FAQ on the wiki in case others need it.

Any idea why MSP over UART was disabled in the first place? I don't know what's gained by doing that.

mikeller, 2017-03-06:
Yes please for adding it to the wiki. This always is forgotten for this sort of change. :-|

It was disabled because it causes problems when users have a serial RX connected to the UART that has MSP enabled by default. In this case the data stream coming from the RX causes MSP to crash the flight controller pretty much immediately, making it impossible to connect to it and disable MSP for that port, without changing the hardware.

FPVlog, 2017-03-06:
I built BF 3.1.6 with MSP enabled on UART1 for all targets. 
You can find the hex files here
https://drive.google.com/folderview?id=0ByGP1-R-32UgUDZhSDJkVWZJdk0

Hope it helps someone ;)

BTW, Joshua Bardwell is gonna release a video tomorrow about that subject;)

quadpaul, 2017-03-09:
Thank you for doing the build on all those targets I know a couple people that need this. And I was just about to build for 3.1.6 for them.

FPVlog, 2017-03-09:
No worries!
Let me know if it works as intended. I've only been able to try it on Omnibus F3

Noctaro, 2017-03-12:
Hi i downloaded the version for CC3D and it worked. Thats great, as i also have a damaged USB on this board. Now i wanted to setup my board using a PWM rx. But had no luck. I tested following setups.

PWM RX
ESC: oneshot125 - PWM separated from PID
only Motor4 working

PWM RX
ESC: oneshot125 - PWM not separated from PID
Motors begin stuttering

PWM RX
ESC: PWM@480
only Motor4 working

PWM RX
ESC: PWM - PWM not separated from PID
only Motor4 working

Setting RX to PPM in Configuration

PPM RX
ESC: PWM - PWM not separated from PID
seems to work

PPM RX
ESC: oneshot125 - PWM not separated from PID
seems to work

PPM RX
ESC: oneshot125 - PWM separated from PID
seems to work

Is there a way to get PWM RX working again, without disabling UART?

mikeller, 2017-03-12:
@Noctaro: Yes, there seems to be a problem right now with PWM on CC3D. Just use PPM for the moment, or upgrade your board (which will also give you access to faster speeds and new features).

Niko-La, 2017-07-20:
@FPVlog I downloaded theNaze32 hex file and uploaded Betaflight for 3.1.6 using local firmware.  Then I set receiver for MSP RX Input.

Is there anything else I must do to send motor commands via USB?

I can get IMU, and Attitude data using the pyMultiWii library and python script but I can not send MSP_SET_RAW_RC commands.

I know the code was working fine with cleanflight in Q42016 firmware.

I can move the motor in the Betaflight configurator.

Any advise?  Also can you please link the Joshua Bardwell video that you mentioned.

thanks

ReganGTX, 2017-08-22:
Can someone please make this happen for Omnibus F4 SD BF 3.2 for me? it's beyond me.

mikeller, 2017-08-25:
@ReganGTX: Try this:

[betaflight_3.2.0_OMNIBUSF4SD.zip](https://github.com/betaflight/betaflight/files/1251150/betaflight_3.2.0_OMNIBUSF4SD.zip)

dyer, 2017-08-31:
Hi mikeller, I was able to get msp over uart working any version < 3.2 compiling from source but it just cannot connect now with any 3.2 release I've tried (BF3 target) although it does seem to build and compile the hex successfully. Have you wizards got any good pointers for me? It's hard to come by anything useful out here in the wild.

Niko-La, 2017-08-31:
@dyer same issue I have. put it aside to clear my mind.  Please let me know if you find a solution.  I tried to roll back to an older version of cleanflight that worked but now the launcher wont accept that.

FPVlog, 2017-09-01:
Here you go guys.
Let me know if it works

https://drive.google.com/file/d/0ByGP1-R-32UgOXQ0akpQMzVQSGs/view?usp=sharing

mikeller, 2017-09-01:
@dyer: In order to get MSP on the first UART, you have to build with `make <target name> OPTIONS="USE_MSP_UART"`.

dyer, 2017-09-01:
That's exactly what I've been doing mikeller, at least I think so? any source I compile for 3.2, it won't let me connect. I get a serial timeout error or some such.

docker run -e "PLATFORM=BETAFLIGHTF3" -e "OPTIONS=USE_MSP_UART" --rm -it -v //c/ETC

I can compile the hex and flash it, but can't connect. I've compiled last working version 3.1.7 iirc and I can flash and connect via ftdi no probs. I'm sure you've got bigger fish to fry. I'll try FPVlogs fw in the morn.

mikeller, 2017-09-01:
Ah I see what you are trying to do. Unfortunately, at the moment, the only option that can be supplied to the docker build is `PLATFORM=<target>`. But adding support for `OPTIONS` is a good suggestion, I can do that later on today.

dyer, 2017-09-01:
Thank you! ..are you a dev on docker/bf as well? My head was going around in circles for a bit after reading your last reply. I don't code in any sense of the word, just trying to figure stuff out and make it work. Getting docker to work was a feat in itself. Thanks for your time mikeller.

mikeller, 2017-09-01:
@dyer: Try a pull for latest on `betaflight/betaflight-build` now, and then try to build with `-e "OPTIONS=USE_MSP_UART"` again.

AndersHoglund, 2017-09-01:
Please note that is is not possible to only add a build `OPTION=USE_MSP_UART` for all targets. There might be conflicts on UART1. Not sure if this is the case for this OMNIBUS target, but it is a problem with BETAFLIGHTF3 for example. I had to create a variant/sub-target BETAFLIGHF3_MSPUART in a branch of my own to be able to do this. Conflict with SBUS_TELEMETRY_UART had to be resolved. 
Checkout my variant here: https://github.com/AndersHoglund/betaflight/blob/bff3_msp_uart_3.2/src/main/target/BETAFLIGHTF3/target.h

dyer, 2017-09-02:
Hey guys thanks for the help but I haven't been able to get it working, tried a clean build with docker and then using Anders modified target. Have you got a build handy I might be able to try Anders? _Maybe_ I'm missing something really obvious here. Btw, it seems you disabled both serial ports, any particular reason for that?

AndersHoglund, 2017-09-02:
Get the build artefacts here: http://andwho.sytes.net:8080/job/AndWho_Betaflight/
 
Removing Softserials was only to make the binary smaller, it did not fit in the flash at some point. But now it seems to do that, so I might just re-enable them again. Not important as I do not use them.

mikeller, 2017-09-03:
@AndersHoglund: `make BETAFLIGHTF3 OPTION=USE_MSP_UART` seems to work just fine for me. I don't have the hardware to test it, but if there is a problem running this firmware and using MSP on UART1, then can you please submit a fix for it (either removing the offending target config when building with `USE_MSP_UART`, or throwing a build error, so as to not frustrate users trying to use it.

mikeller, 2017-09-03:
@dyer: Did you try to pull the `betaflight/betaflight-build:latest` docker image first?

AndersHoglund, 2017-09-03:
@mikeller, Sure compiling for MSPUART like that works just fine. But have you tried running the code on target? Fails. Take a look at the content of `target.h` and you will see the conflict on UART1:
`#define SBUS_TELEMETRY_UART     SERIAL_PORT_USART1`
vs
`#define USE_MSP_UART` 
As you know, `USE_MSP_UART` is hardcoded for UART1 in `src/main/io/serial.c`

I can not see any general solution to this, other than leaving UART1 free and unassigned in all targets. But that would be a pitty for sbus users.....

mikeller, 2017-09-03:
@AndersHoglund: As I said, I don't have the hardware to test the BETAFLIGHTF3 target, and until you stated what exactly was the problem I did not know how to look for other targets with the same problem, in the hope of finding one that I can test with.

I got a fix in #4052, can you test please?

AndersHoglund, 2017-09-03:
OK, that is one solution I did not think of, changing priority(order. Good. Works fine.

But there is a small issue I do not like when building like this: same name, same version, different content. But that is another problem, not directly related to that PR nor this issue,  builds based on command line options not under strict version control. 
Also, for builds like this to work, you MUST start by doing a  `make <target>_clean` as the build system does not re-build automatically when changing command line options. Yet another issue when building with cmd line options, not directly connected here. Just as a hint. 

So I think I will keep a private `BETAFLIGHTF3_MSPUART` target anyhow,  to be able to separate it from the normal `BETAFLIGHTF3`. But that's just me and my preferences.

dyer, 2017-09-04:
Fix worked perfectly, thanks a lot to you both! Great work.

gutster, 2017-09-09:
hey can someone compile me 3.2.0 for the OmnibusF3? Would be super nice since Im too dumb for doing this

dyer, 2017-09-09:
Hi gutster, give this one a go.. https://www.mediafire.com/file/00s4n128zd74hi7/betaflight_3.2.0_OMNIBUS.hex

gutster, 2017-09-09:
@dyer thank you very much worked fine

dyer, 2017-09-10:
Don't thank me, thank the lords above.

clemtrinc, 2017-09-16:
How did you flash it using an ftdi on UART1 ? I just brok my omnibus f3 usb and I'm using Inav and had MSP disabled on UART1. SO right now I can't flash anything except with an STLink according to youtube. It would really help if I can flash it directly with an ftdi since Inav has MSP enabled by default.

Thanks

mikeller, 2017-09-16:
@clemtrinc: Flashing on UART1 with FTDI is never a problem, just use the boot button to get into the boot loader mode, and then flash. But if iNav doesn't have MSP enabled on UART1 by default, you're out of luck for configuring after flashing, unless you build a custom version.

clemtrinc, 2017-09-16:
Yeaaaaaah, I just flashed it no problem on betaflight, it doesn't work on Inav for some reason, like Inav will not talk to bootlaoder somehow, but since reading the OP's solution I tried with betaflight and it worked. Then I tried uploading the latest release of Inav with BF's flasher and it worked !
Just saved a board and an ST-Link despite some guy on youtube claiming you had to connect to the clock pads and all

ReganGTX, 2017-09-16:
https://www.youtube.com/watch?v=5VmhoyL6Z7Q

vicewize, 2017-09-16:
Is there anyway possible you could make a matek f405 build, just broke usb and to get it to click unstable release candidates (just fyi). Im away from computer otherwise I would do it myself. Thanks.

mikeller, 2017-09-16:
@vicewize:

[betaflight_3.2.0_MATEKF405.zip](https://github.com/betaflight/betaflight/files/1308225/betaflight_3.2.0_MATEKF405.zip)

AndersHoglund, 2017-09-16:
I have a few _MSPUART builds running here:  http://andwho.sytes.net:8080/job/AndWho_Betaflight/
Those are usually rebased and in sync with master, but this is done manually at irregular intervals. May not be quite uptodate at all times.

vicewize, 2017-09-16:
Thank you @ mikeller you rock man.

clemtrinc, 2017-09-16:
@ReganGTX   yes that's the video I'm talking about, I don't underdtand why he says you have to use an stlink with the stm software to flash it, since I was able to do it just via uart1 with ftdi on my USB VCP board.

michael-blue2, 2017-10-06:
sorry for asking this :) but how do you flash using uart1? i got a spare lux fc which i wanted to have as a backup. i am on 3.0.1 and everything is working fine with ftdi on uart1 and msp enabled. i want to flash my fc to something earlier like 3.1.6 for now as i have the hex file for my fc with the msp enabled. But HOW do you flash it? do you need st-link? if yes my board doesnt have swd or swdclk for flashing with the st-link. i have tried betaflight(both with bootloader mode and without) and stm loader. still no luck. i always get no response form boatloader. any ideas? i would really appreciate it thanks a lot

clemtrinc, 2017-10-06:
Hi, if it's an omnibus f3 forget about the youtube video just posted stating you need an ST-link, I know you don't because I did it myself.
Even with msp disabled, I was able to flash it via uart1, you need to flash a custom betaflight version compiled with msp enabled on uart1; there's a google drive with almost all the versions.

If you can't flash it, try with other UART or with cleanflight's flasher ( I was initially trying to flash with Inav and it didn't work, I had no bootloader communication) but it worked on BF.

So anyone saying you need an STLINK for the omnibus F3; IT IS NOT TRUE, I'm the living proof of it

jellescope, 2017-10-18:
Hi I have an OMNIBUSF4SD with a broken usb. 
I used the docker container approach with -e "PLATFORM=OMNIBUSF4SD" -e "OPTIONS=USE_MSP_UART". It all compiles and links perfectly. Flashing to the board works fine. The betaflight software is running on the board (ESC do their normal beeping). BUT if I want to connect I get the error: No configuration received within 10 seconds, communication failed. 
I also tried the following:
I changed the OPTIONS line in the makefile directly again no connection possible
I added directly a #define USE_MSP_UART in the target.h file for the OMNIBUSF4SD rebuild, not working
I added the following line:
serialConfig->portConfigs[i].functionMask = FUNCTION_MSP;     directly into the serial.c file (I know not the way to do it). Even that did not work.

@mikeller I used your post for the 3.2.0 OMNIBUSF4SD.hex file from your 25 aug post(at the top of this thread) and this file works fine! So the board is working. 

Why can't I get this working???? I want to be able to compile also the next software releases.
Help is highly appriciated....

mikeller, 2017-10-18:
@jellescope: Are you flashing the new binary with 'Full flash erase' enabled? If not, it might not trigger a config reset, and the changed default settings might not be applied.

jellescope, 2017-10-18:
@mikeller I used the Flash Loader Demonstrator from STline to flash with a global erase. So I assume this should clean it all? 
I used the same settings with your file and that did work perfectly.

KnutKowalski, 2017-10-19:
Hello Guys.
Can someone please help me?
I have an Omnibus F4 Pro with SDCard and broken USB-Port.
I already flashed the Betaflight 3.1.6 with msp on UART1.
Like in the video from Project Blue Falcon (https://www.youtube.com/watch?v=5VmhoyL6Z7Q) shown.
Everything worked fine.

Now i want the newest build 3.2.1 but ofcourse with the msp on UART1 again.

Can someone can explain for an complete noob how to do it?
I use Windows 10 and installed the docker toolbox.

I tried some of the commands and docker downloaded some Betafligt files.
When i try to "build" the hex i get this:
$ docker run -e "PLATFORM=ONMIBUSF4SD" -e "OPTIONS=USE_MSP_UART" --rm -ti -vpwd:/opt/betaflight betaflight/betaflight-b
uild
make: *** No rule to make target 'clean'.  Stop.

What that means?
I have no clue about programming.

Btw. i dont want a build 3.2.1 from some user.
I want to try it for myself.

Thank you! :-)

mikeller, 2017-10-19:
@KnutKowalski: Have a look at the instructions for Windows at https://hub.docker.com/r/betaflight/betaflight-build/.

mikeller, 2017-10-19:
@jellescope: Ok, turns out it was my bad, and the docker container never properly honoured `OPTIONS` supplied via the command line.

It's fixed in betaflight/docker-betaflight-build#4, which is building on dockerhub as we speak. Pull latest on the docker image in 15 or so, and it should work.

jellescope, 2017-10-19:
@KnutKowalski I assume you are using Docker for WIndows. In that case you have to use normal windows style pathnames. Something like this:
docker run -e "PLATFORM=OMNIBUSF4SD" -e "OPTIONS=USE_MSP_UART" --rm -it -v c:/Users/<user>/<subdir>/<subdir>/betaflight:/opt/betaflight betaflight/betaflight-build.
Make sure you also set your drive as shared in the DOcker for Windows settings.
See also the responce from mikeller that you have to get the latest docker-betaflight-build#4 otherwise the OPTIONS will not work. Hope this helps

jellescope, 2017-10-19:
@mikeller Excellent I hope this will work. I am at work now but will try tonight. Fingers crossed. 
By the way I took a look in the Docker file. Are you sure this part of the code is correct: 
    EXTRA_OPTIONS="" && \
    **if [ -n ${EXTRA_OPTIONS} ]; then \**
        EXTRA_OPTIONS="OPTIONS=${OPTIONS}"; \
    fi && \
Should it not be: if [ -n ${OPTIONS} ]; then

I am not an expert in this, so I assume your right.

jellescope, 2017-10-19:
Hi Michael,

Thanks for the fast response on my request I am new to this github system so hope this response will reach you.
I want to share what I did on my personal copy of the docker build image. It is a little too much for the  thread I think 😊

In the docker file I changed the following part of the code:

CMD if [ -z ${PLATFORM} ]; then \
        PLATFORM="NAZE"; \
    fi && \
    EXTRA_OPTIONS="" && \
    if [ -n ${OPTIONS} ]; then \
       EXTRA_OPTIONS="OPTIONS=${OPTIONS}"; \
    fi && \
    if [ ${PLATFORM} = ALL ]; then \
        make ARM_SDK_DIR=${ARM_SDK_DIR} clean_all && \
        make ARM_SDK_DIR=${ARM_SDK_DIR} all ${EXTRA_OPTIONS}; \
    else \
        CHECK="make ARM_SDK_DIR=${ARM_SDK_DIR} TARGET=${PLATFORM} ${EXTRA_OPTIONS}" && \
        echo ${CHECK} && \
        make ARM_SDK_DIR=${ARM_SDK_DIR} clean TARGET=${PLATFORM} && \
        make ARM_SDK_DIR=${ARM_SDK_DIR} TARGET=${PLATFORM} ${EXTRA_OPTIONS}; \
    fi

During the run I get the following response from the echo command:
make ARM_SDK_DIR=/usr/ TARGET=OMNIBUSF4SD OPTIONS=USE_MSP_UART
This seems correct? Do you agree?

By the way I am building version 3.2.2 from the 3.2.x maintenance branch.
The build result is at least different from my previous attempt:
Failed hex file was:
   text    data     bss     dec     hex filename
279242    9696   46996  335934   5203e ./obj/main/betaflight_OMNIBUSF4SD.elf

The new hex file is:
   text    data     bss     dec     hex filename
279314    9696   46996  336006   52086 ./obj/main/betaflight_OMNIBUSF4SD.elf

Will try to load the file tonight and check if

Thanks for the support really helpful.

Kind regards,
Jelle Visser.

jellescope, 2017-10-19:
@mikeller Sorry, I see the email response will go directly to the thread. Cool system :-)

KnutKowalski, 2017-10-19:
Ok guys.
The command docker run -e "PLATFORM=OMNIBUSF4SD" -e "OPTIONS=USE_MSP_UART" --rm -it -v c:/Users/MYNAME/betaflight:/opt/betaflight betaflight/betaflight-build worked.

But when i try to connect via the ftdi adapter to the betaflight chrome app (like with the older 3.1.6, that connected with no problem) it just dont work.
Btw,. on the omnibus normally the green led is on and the blue led blinks.
Now only the green led is on...

KnutKowalski, 2017-10-19:
`thoma@DESKTOP MINGW64 ~
$ docker run -e "PLATFORM=OMNIBUSF4SD" -e "OPTIONS=USE_MSP_UART" --rm -it -v //c/Users/thoma/betaflight:/opt/betaflight
 betaflight/betaflight-build
Cleaning OMNIBUSF4SD
rm -f ./obj/betaflight_3.3.0_OMNIBUSF4SD.bin ./obj/betaflight_3.3.0_OMNIBUSF4SD.hex ./obj/main/betaflight_OMNIBUSF4SD.elf ./obj/main/OMNIBUSF4SD/startup_stm32f40xx.o ./obj/main/OMNIBUSF4SD/target/system_stm32f4xx.o ./obj/main/OMNIBUSF4SD/drivers/accgyro/accgyro_mpu.o ./obj/main/OMNIBUSF4SD/drivers/adc_stm32f4xx.o ./obj/main/OMNIBUSF4SD/drivers/bus_i2c_stm32f10x.o ./obj/main/OMNIBUSF4SD/drivers/dma_stm32f4xx.o ./obj/main/OMNIBUSF4SD/drivers/inverter.o ./obj/main/OMNIBUSF4SD/drivers/light_ws2811strip_stdperiph.o ./obj/main/OMNIBUSF4SD/drivers/pwm_output_dshot.o ./obj/main/OMNIBUSF4SD/drivers/serial_uart_init.o ./obj/main/OMNIBUSF4SD/drivers/serial_uart_stm32f4xx.o ./obj/main/OMNIBUSF4SD/drivers/system_stm32f4xx.o ./obj/main/OMNIBUSF4SD/drivers/timer_stm32f4xx.o ./obj/main/OMNIBUSF4SD/drivers/accgyro/accgyro_mpu6500.o ./obj/main/OMNIBUSF4SD/drivers/accgyro/accgyro_spi_mpu6000.o ./obj/main/OMNIBUSF4SD/drivers/accgyro/accgyro_spi_mpu6500.o ./obj/main/OMNIBUSF4SD/drivers/barometer/barometer_bmp085.o ./obj/main/OMNIBUSF4SD/drivers/barometer/barometer_bmp280.o ./obj/main/OMNIBUSF4SD/drivers/barometer/barometer_ms5611.o ./obj/main/OMNIBUSF4SD/drivers/compass/compass_hmc5883l.o ./obj/main/OMNIBUSF4SD/drivers/max7456.o ./obj/main/OMNIBUSF4SD/./lib/main/DSP_Lib/Source/BasicMathFunctions/arm_mult_f32.o ./obj/main/OMNIBUSF4SD/./lib/main/DSP_Lib/Source/TransformFunctions/arm_rfft_fast_f32.o ./obj/main/OMNIBUSF4SD/./lib/main/DSP_Lib/Source/TransformFunctions/arm_cfft_f32.o ./obj/main/OMNIBUSF4SD/./lib/main/DSP_Lib/Source/TransformFunctions/arm_rfft_fast_init_f32.o ./obj/main/OMNIBUSF4SD/./lib/main/DSP_Lib/Source/TransformFunctions/arm_cfft_radix8_f32.o ./obj/main/OMNIBUSF4SD/./lib/main/DSP_Lib/Source/CommonTables/arm_common_tables.o ./obj/main/OMNIBUSF4SD/./lib/main/DSP_Lib/Source/ComplexMathFunctions/arm_cmplx_mag_f32.o ./obj/main/OMNIBUSF4SD/./lib/main/DSP_Lib/Source/StatisticsFunctions/arm_max_f32.o ./obj/main/OMNIBUSF4SD/./lib/main/DSP_Lib/Source/TransformFunctions/arm_bitreversal2.o ./obj/main/OMNIBUSF4SD/drivers/flash_m25p16.o ./obj/main/OMNIBUSF4SD/io/flashfs.o ./obj/main/OMNIBUSF4SD/build/build_config.o ./obj/main/OMNIBUSF4SD/build/debug.o ./obj/main/OMNIBUSF4SD/build/version.o ./obj/main/OMNIBUSF4SD/target.o ./obj/main/OMNIBUSF4SD/config.o ./obj/main/OMNIBUSF4SD/main.o ./obj/main/OMNIBUSF4SD/common/bitarray.o ./obj/main/OMNIBUSF4SD/common/crc.o ./obj/main/OMNIBUSF4SD/common/encoding.o ./obj/main/OMNIBUSF4SD/common/filter.o ./obj/main/OMNIBUSF4SD/common/huffman.o ./obj/main/OMNIBUSF4SD/common/huffman_table.o ./obj/main/OMNIBUSF4SD/common/maths.o ./obj/main/OMNIBUSF4SD/common/printf.o ./obj/main/OMNIBUSF4SD/common/streambuf.o ./obj/main/OMNIBUSF4SD/common/typeconversion.o ./obj/main/OMNIBUSF4SD/config/config_eeprom.o ./obj/main/OMNIBUSF4SD/config/feature.o ./obj/main/OMNIBUSF4SD/config/parameter_group.o ./obj/main/OMNIBUSF4SD/config/config_streamer.o ./obj/main/OMNIBUSF4SD/drivers/adc.o ./obj/main/OMNIBUSF4SD/drivers/buf_writer.o ./obj/main/OMNIBUSF4SD/drivers/bus.o ./obj/main/OMNIBUSF4SD/drivers/bus_i2c_config.o ./obj/main/OMNIBUSF4SD/drivers/bus_i2c_busdev.o ./obj/main/OMNIBUSF4SD/drivers/bus_i2c_soft.o ./obj/main/OMNIBUSF4SD/drivers/bus_spi.o ./obj/main/OMNIBUSF4SD/drivers/bus_spi_config.o ./obj/main/OMNIBUSF4SD/drivers/bus_spi_pinconfig.o ./obj/main/OMNIBUSF4SD/drivers/bus_spi_soft.o ./obj/main/OMNIBUSF4SD/drivers/buttons.o ./obj/main/OMNIBUSF4SD/drivers/display.o ./obj/main/OMNIBUSF4SD/drivers/exti.o ./obj/main/OMNIBUSF4SD/drivers/io.o ./obj/main/OMNIBUSF4SD/drivers/light_led.o ./obj/main/OMNIBUSF4SD/drivers/resource.o ./obj/main/OMNIBUSF4SD/drivers/rcc.o ./obj/main/OMNIBUSF4SD/drivers/serial.o ./obj/main/OMNIBUSF4SD/drivers/serial_pinconfig.o ./obj/main/OMNIBUSF4SD/drivers/serial_uart.o ./obj/main/OMNIBUSF4SD/drivers/serial_uart_pinconfig.o ./obj/main/OMNIBUSF4SD/drivers/sound_beeper.o ./obj/main/OMNIBUSF4SD/drivers/stack_check.o ./obj/main/OMNIBUSF4SD/drivers/system.o ./obj/main/OMNIBUSF4SD/drivers/timer.o ./obj/main/OMNIBUSF4SD/drivers/transponder_ir.o ./obj/main/OMNIBUSF4SD/drivers/transponder_ir_arcitimer.o ./obj/main/OMNIBUSF4SD/drivers/transponder_ir_ilap.o ./obj/main/OMNIBUSF4SD/drivers/transponder_ir_erlt.o ./obj/main/OMNIBUSF4SD/fc/config.o ./obj/main/OMNIBUSF4SD/fc/fc_dispatch.o ./obj/main/OMNIBUSF4SD/fc/fc_hardfaults.o ./obj/main/OMNIBUSF4SD/fc/fc_msp.o ./obj/main/OMNIBUSF4SD/fc/fc_msp_box.o ./obj/main/OMNIBUSF4SD/fc/fc_tasks.o ./obj/main/OMNIBUSF4SD/fc/runtime_config.o ./obj/main/OMNIBUSF4SD/io/beeper.o ./obj/main/OMNIBUSF4SD/io/serial.o ./obj/main/OMNIBUSF4SD/io/statusindicator.o ./obj/main/OMNIBUSF4SD/io/transponder_ir.o ./obj/main/OMNIBUSF4SD/io/rcsplit.o ./obj/main/OMNIBUSF4SD/msp/msp_serial.o ./obj/main/OMNIBUSF4SD/scheduler/scheduler.o ./obj/main/OMNIBUSF4SD/sensors/battery.o ./obj/main/OMNIBUSF4SD/sensors/current.o ./obj/main/OMNIBUSF4SD/sensors/voltage.o ./obj/main/OMNIBUSF4SD/fc/fc_init.o ./obj/main/OMNIBUSF4SD/fc/controlrate_profile.o ./obj/main/OMNIBUSF4SD/drivers/camera_control.o ./obj/main/OMNIBUSF4SD/drivers/gyro_sync.o ./obj/main/OMNIBUSF4SD/drivers/rx_nrf24l01.o ./obj/main/OMNIBUSF4SD/drivers/rx_spi.o ./obj/main/OMNIBUSF4SD/drivers/rx_xn297.o ./obj/main/OMNIBUSF4SD/drivers/pwm_esc_detect.o ./obj/main/OMNIBUSF4SD/drivers/pwm_output.o ./obj/main/OMNIBUSF4SD/drivers/rx_pwm.o ./obj/mai [... cut ...]
%% stm32f4xx_it.c
%% usbd_desc.c
%% usbd_usr.c
%% usbd_cdc_vcp.c
%% serial_usb_vcp.c
%% usb_io.c
Linking OMNIBUSF4SD
/usr//bin/arm-none-eabi-size ./obj/main/betaflight_OMNIBUSF4SD.elf
   text    data     bss     dec     hex filename
 279408    9892   47148  336448   52240 ./obj/main/betaflight_OMNIBUSF4SD.elf
/usr//bin/arm-none-eabi-objcopy -O ihex --set-start 0x8000000 obj/main/betaflight_OMNIBUSF4SD.elf obj/betaflight_3.3.0_OMNIBUSF4SD.hex
make[1]: Leaving directory '/opt/betaflight'

thoma@DESKTOP MINGW64 ~`

[49 more replies]

closed 2017-10-22 by mikeller