betaflight/betaflight #2567: Shut USB VCP port on my flight controller

Jdroid9, 2017-03-05:
Anyone know how I could re-open it?  I have tried to re-flash and have not managed to get it to work.

FPVlog, 2017-03-06:
I think you can reflash the board via UART1 with an FTDI adapter.
That's what I did when my USB port was broken so I think it could work for you as well.

But we need the experts to validate that ;)

mikeller, 2017-03-06:
@Jdroid9 can you be more specific what you mean by 'shut'? If it's a hardware fault, then flashing over UART1 with an FTDI adapter is possible, like @FPVlog says.

If it's just misconfigured, then shorting the boot pins / pushing the boot button when powering up will bring it up in DFU mode, and reflashing with 'Full chip erase' checked will force a valid config.

Jdroid9, 2017-03-06:
What I meant by shut was that I accidentally flipped the switch in betaflight that shut the USB VCP.

mikeller, 2017-03-06:
Ah. That should reset itself on the next reboot, unless you had another port selected for MSP. But anyhow, booting into DFU (boot pin / pad), and reflashing with 'Full chip erase' will fix.

cabecinha69, 2017-09-08:
Hello. Accidently I tuirn off the usb vcp port turn off...damm...and now? Who can I turn on?

Every time I trie to use betaflight, I get port error...dammmm

AndersHoglund, 2017-09-09:
@cabecinha69 Read messages prior to your input. Bootloader mode, (re-)flash with full chip erase, restore saved diff file.

closed 2017-10-23 by jflyper

cgaxtreme10, 2018-08-26:
Dear friends
I have the flight controller matek f722 ... I assigned by softserial the pad pwm for smart port telemetry. Then in ports I gave him save and reboot and after that the controller card is not recognized by the pc. I have reinstalled the drivers multiple times without success. In the device manager, it appears as an unknown serial device and the controller card turns on, initially turns on a red LED, then blinks the blue and red LED and ends up with the red led on permanently without recognizing the controller PC. What do I have to do to restore the controller or for the PC to recognize it? Many thanks. Best regards

jflyper, 2018-08-26:
@cgaxtreme10 
Try re-flashing the board with full-chip-erase.