betaflight/betaflight #2569: omnibusF4 3.2.0-1079 DSHOT not working

hahatou, 2017-03-06:
3.1.6 ONIBUSF4 dshot is normal

jflyper, 2017-03-06:
Can you be more specific about how "not working" ?

hahatou, 2017-03-06:
@jflyper Power on,ESC only self check, no confirm sound

AirBreak69, 2017-03-06:
I can confirm @hahatou's issue on my original REVO.
The confirm sound reappears if configured back to Oneshot125 or Multishot.
I went back to older builds: Build 1067 was the last with working DSHOT on my REVO.
I also tried CLI defaults before activating DSHOT. Even the slower DSHOTs don't work.
Another hint: Beginning with build 1068 the CLI page in the configurator shows garbage characters in the first line.

AirBreak69, 2017-03-07:
Just an update: I tried the latest build 1092 - nothing has changed.

hahatou, 2017-03-08:
@AirBreak69 Build 1067 is last normal?

sleeslee, 2017-03-08:
Build 1068 is last normal on my AirBot F4

AirBreak69, 2017-03-08:
@hahatou, i have to correct: Build 1068 is also working on my REVO.
Build 1069 is the first non working build.
The currently latest build 1097 has the DSHOT failure too.

jflyper, 2017-03-09:
@ here
This issue has been recognized by devs. Will be a fix soon.

AirBreak69, 2017-03-18:
Problem has been addressed in #2662 and seems to be fixed with build 1119.
I tested build 1121 on my REVO - DSHOT is working again on F4s.
Can be closed...

closed 2017-03-19 by mikeller