betaflight/betaflight #2574: Led_strip enabling on original Revo 3.1.6 makes FC completely bricked.

Jeuke26, 2017-03-06:
When I enable led_strip in config page ( no wires connected on the Revo the FC looses connection to BF, and can't be accessed anymore. No com port pop up nothing. Only way to revive is: boot the pads, change drivers in Zadig to CDC serial and back to winusb. Then open in DFU mode in BF and flash back BF 3.1.6. Flashing is corrupted 3, 4 times before the flashing actually is succes full. I can configure the entire board and calibrate esc. But as soon as I enable led_strip the whole process has to be started again.

AndersHoglund, 2017-03-06:
I can not reproduce this. @borisbstyle , did you manage to do that?
But I only have the RTFQ Revo ACRO, not the original. But that should not really matter, or?
All I do is Flash 3.1.6, reset settings, enable LED-strip, save and reboot. No problem reconnecting.

@Jeuke26 , have you tried with only default settings and LED-strip enabled? Does this happen only when you have ".. configure the entire board..." ? Then please post your configuration, using the diff command.

borisbstyle, 2017-03-06:
No i dont have one, but since I saw 2 reports I thought it should be real bug?

Jeuke26, 2017-03-06:
Ok, I'll try this evening by keeping all defaults and then enabling led_strip in config page. The problem is I can't post a dump or error because the Revo completely looses its connection. I have to manually reflash the drivers in Zadig and then put into DFU mode and reflash latest build. 

Philip Dezsö

Jeuke26, 2017-03-06:
The minute i click save I can't post the config because, well the above.

AndersHoglund, 2017-03-06:
Well, do the diff before enabling the LEDs then. I assume it is working then.
Oh, by the way. You do not have to mess with Zadig every time. Once the WinUSB drivers are properly installed for the STM32 BOOTLOADER device you are all set. Only need to be done once.

Jeuke26, 2017-03-06:
Unfortunately it doesn't. The winusb is installed when the Revo is in bootloader mode but the BF Configurator isn't able to flash the Revo. It hangs on the flash part and nothing happens. Only removing and reinstalling the winusb lets BF flash the build.

Jeuke26, 2017-03-06:
Entering CLI Mode, type 'exit' to return, or 'help'

# diff

# version
# Betaflight / REVO 3.1.6 Feb 21 2017 / 14:37:31 (1fd502c)

# name

# resources

# mixer
mixer QUADX1234

# servo

# servo mix

# feature

# beeper

# map
map TAER1234

# serial
serial 0 64 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1700 2100

# adjrange

# rxrange

# rxfail

# master
set min_throttle = 1025
set motor_pwm_protocol = MULTISHOT
set disarm_kill_switch = OFF
set serialrx_provider = SPEK2048
set pid_process_denom = 1

# profile
profile 0

rateprofile 0

# rateprofile
rateprofile 0

Jeuke26, 2017-03-06:
no wires connected. all is ok, but when i enable LED_Strip the Revo goes bananas

AndersHoglund, 2017-03-06:
Still can not reproduce the problem even with your config added. Both with nothing connected to a naked Revo on the bench, and with LED-strip connected. Powered from USB only and from extrenal 5V. Do I understand you correctly that it all works OK for also in the case where "..no wires connected. all is ok" ? So it fails only when  LED-strip is connected and enabled.? Or is it some other wires??

I think I also read on RCG that you had some other symptoms when connecting a battery, true?
That might lead to a suspicion you have a problem with power supply to the Revo. A LED-strip can draw quite some current causing a brown-out. Have you measured voltage on the +5V rail with and without LED-strip connected and enabled?

Can do much more, sorry.  Some one else with a fully populated original Revo might give it a try to reproduce the problem.

Jeuke26, 2017-03-07:
The problem arises with or without wires connected to the Revo. The initial issue with the FC loosing connection to Pc upon lipo connection was due to a ground loop. 

So, I can change whatever I want in BFC, no wires except USB connected. As soon as I enable led_strip ( nothing connected on pad 5 ) the Revo goes into some kind of brick mode. If it's is a hardware issue which always possible. Maybe somebody could take over and chime in. Thanks.

Arakon, 2017-03-09:
I can at least tell you that with the airbotF4 target on the banggood revo clone with buzzer/LED support, nothing like this happens.. so it may be specific to your hardware. Try flashing airbot target and see if it still does it.

Jeuke26, 2017-03-09:
Good Point, i will give that a try and report back.

Jeuke26, 2017-03-10:
ok, this is what has been tried. the Airbot target gives the same error reproduction. So i took a second Revo out of a fully racing frame and undid al wiring to have a blank FC. with this one, it takes the led_strip command without issue. so i copied the exact same way to the perfectly working Revo ( did many flights on it with BF 3.1.X ) and again it got "bricked". there is no damage whatsoever, no crash, no water, no shorts nothing. So if anybody has a clue why a pure software command bricks a fully working Revo. ALL other settings can be changed altered or disabled, except "LED_STRIP" command.....

Jeuke26, 2017-03-10:
I don't know if this issue may be closed but i ran the full board with a continuity test I noticed that signal pad 5 is somehow shorted to all other pads ( only pad 5 ) so enabling LED_STRIP must engage some kind of fixed signal once enabled causing the board to go haywire.

closed 2017-03-11 by mikeller