betaflight/betaflight #2576: Serial port is rebooting after throttle 100% on FlySky

Tverdokhleb, 2017-03-06:
Set:
- Transmitter FlySky FS-I6;
- Receiver FS-A8S;
- F3 V4 AIO flight controller;
I have flashed BetaFlight 3.1.6 (SPRACINGF3, default parameters), configured receiver via I-BUS protocol.
Steps to reproduce a bug:
1. Set throttle to 100% (Should be shown 2000 in Receiver Tab);
2. Set pitch to 100% (Should be shown 2000 in Receiver Tab);
After those steps serial port is rebooting and it is blinking in configurator menu.
---
I have also reproduced those steps on FlySky FS-I6, receiver FS-IA4B and BeeRotor F3 flight controller.

Tverdokhleb, 2017-03-06:
This bug was also reproduced on RadioLink AT9.

mikeller, 2017-03-06:
Could it be that you have stick arming enabled and this is just the reaction of the flight controller to a stick command?

Tverdokhleb, 2017-03-07:
Maybe. It is also reproduced during flytime (FC freezes).

mikeller, 2017-03-07:
'flytime' as in while the FC is armed? If that's the case then it's not stick commands.

Tverdokhleb, 2017-03-07:
Yes, I.e you can fly.

Tverdokhleb, 2017-03-07:
max_check parameter by default is set as 1900. If set throttle stick from 1900 and pitch/roll till 2000, then FC is freezes on several seconds (Beeper beeps if batery is connected). 
We found one of the solution - set max_check as 2000 and set throttle stick value in transmitter less than 2000. But it is temporary solution.

mikeller, 2017-03-07:
Ok, this makes it even weirder. `max_check` is only used to detect stick commands. Can you try if the problem goes away when you change to switch arming (i.e. define a range for 'ARM' on the 'Modes' tab?

Tverdokhleb, 2017-03-07:
ARM mode is not affected on the problem. The problem is reproduced in armed/dissarmed mode.

mikeller, 2017-03-07:
Ok, in that case I can't help you any more, since I don't have the hardware to reproduce.

AndersHoglund, 2017-03-07:
Weird, but you are most probably entering a stick command that causes the FC to reboot or crash.
https://github.com/cleanflight/cleanflight/raw/master/docs/assets/images/StickPositions.png

Tverdokhleb, 2017-03-07:
It is not a bug. It is a result of Stick command.

closed 2017-03-07 by Tverdokhleb

gpwelter, 2017-04-21:
Not a bug or result or stick command. The rx sucks. I switched it out with a different model rx and it behaves correctly without work arounds. Prior to drilling down to the rx. I switched out everything except the motor and frame. The work around works to an extent but not always, causing impact with anything in it's path. Sorry Boris and Hydra any many of the other developers. I believe 95% of the issues is jankiness of the hardware setup and not your genius. You + quads = FPV Love!